fritzing/fritzing-app #3173: Some issue in fritzing-app for mac os x.

imeteora, 2015-12-22:
1, Current stable version of fritzing-app for mac os x maybe was using the old version Qt, the right-click menu(quick menu) display in wrong place. I had check the fritzing-app in my own mac with latest Qt (Qt 5.5, I check the Qt offical website, the newest version is 5.6). The quick menu show in the right place, so I hope the newest fritzing-app will be out soon.

2, The deploy shell script, called `deploy_fritzing_mac.sh`, download the whole fritzing-app source tree every-time when it starts. The whole compile speed had been slowed down by this git clone progress. I replace git clone code in Line 26 instead of `cp -f -R ./* $deploysrcdir `. Now, speed up, no network condition needs. The whole resource files are here, why clone it from git respository again?

aknoerig, 2016-01-26:
Thanks, we're indeed waiting for the Qt 5.6 release, which is a long-term support release, to publish 0.9.3. We will then also update the release script, since several other things got changed about it, too.

aknoerig, 2016-01-28:
Fixed in 0bf6109, about to be merged.

closed 2016-01-28 by aknoerig