fritzing/fritzing-app #3177: Mouse handles should remain the same size across zoom levels

deshipu, 2016-01-15:
This is a usability observation, and I think following this could greatly improve the ease of use for Fritzing.

Currently, when we zoom into the PCB view (and possibly other views too), all the mouse handles (trace benpoints, footprint outlines, etc.) zoom too. This means that zooming doesn't help with working with very tight designs, where those handles overlap. If instead the handles remained a constant size (large enough to conveniently click, small enough to not obscure too much), then zooming in would result in them being displayed farther apart, and thus easier to manipulate accurately. You can see how GIMP and Inkscape handle that, and how that makes it much easier to work both at large and small zoom levels.

idontusenumbers, 2018-01-16:
This times a hundred. I regularly find it absolutely impossible to select a wire end point, drop it on the appropriate connector, move a component, etc.

KjellMorgenstern, 2024-09-23:
There are (or hopefully, where) multiple things to fix for this issue.

All screenshots taken at zoom level 3000% . Most show an RP2040 (QFN part) , wires are 8mil
Comparing 1.0.3 to the upcoming 1.0.4.

### 1. Ratsnest lines getting in the way. 

_Fritzing 1.0.3 ratsnest lines with way to big hitbox_
[image]
The Hitbox for ratsnest line obstructs wire end. Note: I modifed 1.0.3 to show the hitbox. A vanilla Fritzing would give no indication that a click will grab the ratsnest line. Annoying.

_Fritzing 1.0.4 ratsnest lines don't increas with zoom_
[image]
The hitbox of the ratsnest line is only slightly bigger than the line itself. It is now shown with a hover shadow by default. Ratsnest and their hitbox keep the same absolute widht accross all zoom levels.

### 2. Dotted selection markers not helpful on zoom

_Fritzing 1.0.3 selection markers are irritating at best_
[image]

_Fritzing 1.0.4 selection markers are relative to the trace width_
[image]

### 3. Mouse handles clutter view.
Did you spot that there is a unnecessary bendpoint between the 6th and 7th connector of the row?
Overlapping mouse handles often mean that there are more hitbox candiates than useful. Of course it is always the wrong item that takes priority.

_Fritzing 1.0.3 bendpoint hard to spot, mouse needs to be at a seemingly random spot far away to hit it_
[image]

_Fritzing 1.0.4 directly however the bendpoint
[image]

_and remove it with double click to get a straight wire. Fritzing 1.0.3_
[image]

_1.0.4 ..._
[image]

KjellMorgenstern, 2024-09-23:
### 4. Wires easily jump to nearby connectors

_Fritzing 1.0.3 detaching the wire from VDD (7th pad of the row)_
[image]
_but it directly snaps to the nearby bendpoint_ :-1: 
[image]

_Fritzing 1.0.4 grab the wire_
[image]
_and drop it at an empty place to detach it_
[image]

KjellMorgenstern, 2024-09-23:
### 4b. Wires easily jump to nearby connectors

Same effect as 4a, but quite a different part of the code, so worth to mention. 
Connecting the lose wire to a connector.

_Fritzing 1.0.3. Where will it snap?_
[image]
_... adding some insult, the hover text already told it will use this connector _
[image]

_Fritzing 1.0.4 even with shaky hands_
[image]
_probably the correct connector. Using the distance to the candidates for a guess what the user wanted_
[image]

*) If there are multiple candidates, we still prioritize by layer, but we now also take the distance into account.
We don't need to rely on the layers as much anymore. This will probably enable some more improvements for edge cases, e.g in the breadboard view.

closed 2024-10-10 by KjellMorgenstern