fritzing/fritzing-app #3178: change standart grid to 0.05inch

el-j, 2016-01-20:
for pcb production this grid makes more sense. often failures happen because of the 0.1inch grid.
should be changed in new releases

ebecheto, 2019-04-04:
In my opinion, this should be a parameter that you could be modified by a user, not a fixed value.

failiz, 2021-12-14:
> In my opinion, this should be a parameter that you could be modified by a user, not a fixed value.

This can already be changed by the user. 

I am not against changing the default grid settings and I think that it makes sense but I cannot see how this can cause failures.

KjellMorgenstern, 2021-12-14:
Not really failures in the sense of software bugs, but against the grid (or without the help of a 0.05 grid),
maybe some users route connections like below. If you keep the grid active, it is really hard to route a thin wire
between the connectors.

[image]

With a 0.05 grid, it looks much better. But even with a super fine wire, the DRC will complain at standard settings. Not sure if this is justified, or if the default DRC should allow this:
[image]