fritzing/fritzing-app #3179: change properties of multiple parts of the same type/class/property

el-j, 2016-01-20:
this function is really needed.

examples: 
 - select all wires and change there size
 - choose all via's and change drill size

etc.

M4nuski, 2016-01-30:
so far it works only with [select all wires] but not [select all vias] or custom selection.
Only change the property of the "first" selected item.

DrItanium, 2020-01-26:
I'll look into this but no promises, this issue has added so much time to some of my projects. Odd that it doesn't affect PCB trace widths though...

failiz, 2024-01-08:
duplicate of https://github.com/fritzing/fritzing-app/issues/3620
The selection of multiple traces in PCB and wires in BB allows to change the width or color of all of them. Vias properties cannot be changed in bulk.

KjellMorgenstern, 2024-01-08:
I changed the title, so we can focus on the vias in the issue.

KjellMorgenstern, 2024-01-08:
[vias.fzz.zip](https://github.com/fritzing/fritzing-app/files/13862004/vias.fzz.zip)

[image]

To select all vias:
<kbd>Ctrl+F</kbd> + Enter "Via" , or choose select all Vias from the routing menu.

KjellMorgenstern, 2024-01-08:
Changing properties ( or showing common properites ) of the selection is not possible. Only the part shown in the inspector will be changed, or the selection will focus on one part on right click.