fritzing/fritzing-app #3180: ICs shifted and disconnected upon load

etaba, 2016-01-20:
My schematic contains two generic 12 pin ICs. Every time I load the project they are disconnected from their wires, shifted to the left, and the pinout object (containing the IC's pin labels and name) floating elsewhere. Only happens for the 12 pin ICs, whereas my switches, transistors, microcontroller, etc all load fine.
Heres what it looks like when I load:

[image]

aknoerig, 2016-01-26:
Seems to be an issue with the part. Which exact part are you using here?

etaba, 2016-01-26:
I just used the generic IC, which had 8 pins i believe, then in the options I changed the number of pins to 14. Should I use a different part for this? In retrospect, perhaps I should have selected SOIC-14 package instead of using DIP and changing the pin count.

Here is a pic of the Part Inspector when I click on it:
[image]

Otherwise I very much enjoy the product

KjellMorgenstern, 2019-11-20:
Can this still be reproduced? If yes, please attach an fzp file.

oxygene, 2020-01-27:
Just happened to me with a generic IC, set to 16 pins and rotated by 90 degrees.
I then saved the part into "my parts" and exported it from there. See attachment. Happened with v0.9.4.

[DIP - 16 pins.zip](https://github.com/fritzing/fritzing-app/files/4119533/DIP.-.16.pins.zip)

closed 2023-07-14 by KjellMorgenstern