fritzing/fritzing-app #3182: There

FS-rcarroll, 2016-01-28:
This is the same as this archived problem:
http://fritzing.org/forum/thread/1375/

When I open I get this error: http://imgur.com/Rv3Benj

I'm running Mac El Capitan (but this happened in Mountain Lion, as well)

aknoerig, 2016-01-28:
Hmm, still a weird problem. Did you try a fresh installation, also wiping your use folder at ~/.config/Fritzing ?

FS-rcarroll, 2016-01-29:
I did try a fresh install. I took a look at the ~/.config folder in my user directory but it is empty. Am I looking in the right place?

aknoerig, 2016-01-29:
Yes, but it could mean that Fritzing isn't able to create a user folder there. Maybe a permissions problem, as Jonathan describes in the last forum post. You could try the two ideas he mentions there.

FS-rcarroll, 2016-01-29:
Would the /.config file be in my user directory or in the var or usr directory?

closed 2016-01-30 by FS-rcarroll