fritzing/fritzing-app #3185: Gerber files contain weird sections

markthepiggy, 2016-02-04:
There is s problem with the way Fritzing generates gerber files for some components. 

Rather than smooth lines, the pads of some components are constructed with lots of little steps. The issue can be seen in the following screen shots taken from Online Gerber viewer (http://www.gerber-viewer.com/default.aspx).

This is the board showing the field of view:

[image]

and here is the enlarged section of interest:

[image]

You can see that the pad on the bottom is made up of a smooth arc. This cuts very quickly on a cnc mill. The track is also ok. The top pad is made up of lots of jagged lines. This cuts very slowly.

The bottom pad is from the component 'RJ45 Jack'. The top pad is from the component 'Generic Female Header'. 

I also exported the board as an svg. The svg shows no sign of the problem:

[image]

I have not been able to find any reports of this problem using Google or in the Fritzing bug tracker.

I am using Version 0.9.2 on windows 10.

markthepiggy, 2016-02-04:
This issue looks like it has been addressed by an outstanding pull request byt nraynaud on May 13, 2015. There are obviously some issues with the pull request, but it would be great to get these fixed and incorporate the changes.

KjellMorgenstern, 2019-11-20:
Unfortunately the gerber export pull request had to be reverted, so this issue probably still exists.