fritzing/fritzing-app #3186: SVG export reverses/flips the text

zeehio, 2016-02-05:
When exporting as SVG, the text is flipped and or rotated when it should not be. Here is a simple example that can be used for testing:

## SVG export (incorrect, converted to PNG):
[image]

## PNG export (correct):
[image]

## SVG figure to see the generated "svg code"

SVG figure (In a zip file, so it can be attached in a github issue)
[bug_export_label_rotated_schem.svg.zip](https://github.com/fritzing/fritzing-app/files/118922/bug_export_label_rotated_schem.svg.zip)

## Fritzing test

FZZ file (with zip extension added, so it can be attached in a github issue):
[bug_export_label_rotated.fzz.zip](https://github.com/fritzing/fritzing-app/files/118913/bug_export_label_rotated.fzz.zip)

zeehio, 2016-02-05:
Additionally, the colors are not converted to the black silkscreen and "#F7BD13" standard copper colors. This makes some parts appear with white silkscreen, hard to distinguish if there is a white background (e.g  https://github.com/fritzing/fritzing-parts/blob/master/svg/core/pcb/connector_power_pj202ah.svg)

KjellMorgenstern, 2019-11-20:
Easy to reproduce with current 0.9.5 develop (and probably all other versions)

KjellMorgenstern, 2019-11-21:
Possibly related to #2772

closed 2023-06-23 by KjellMorgenstern