fritzing/fritzing-app #3188: Could not open any SVG in Part Editor

mediatainment, 2016-02-08:
I wann add a 7-segment-display part, by editing the original one and save it with a different name.

When I open a new image, the fritzing app gives me an error, but in the most cases it crashes.

Am I doing something wrong?

best regards

Screenrecord:

[image]

my tried svg:

```

<?xml version="1.0" encoding="utf-8"?>
<!-- Generator: Adobe Illustrator 18.0.0, SVG Export Plug-In . SVG Version: 6.00 Build 0)  -->
<!DOCTYPE svg PUBLIC "-//W3C//DTD SVG 1.1 Basic//EN" "http://www.w3.org/Graphics/SVG/1.1/DTD/svg11-basic.dtd">
<svg version="1.1" baseProfile="basic" id="Layer_1" xmlns:svg="http://www.w3.org/2000/svg"
	 xmlns="http://www.w3.org/2000/svg" xmlns:xlink="http://www.w3.org/1999/xlink" x="0" y="0" viewBox="0 0 114.3 38.9"
	 xml:space="preserve">
<g>
	<g id="breadboard">
		<circle id="connector0pin" fill="none" stroke="#9A916C" stroke-width="0.381" cx="31.9" cy="33.8" r="2"/>
		<g id="connector0terminal">
		</g>
		<path fill="#9A916C" d="M3.4,28.8c1.4,0,2.5,1.1,2.5,2.5v5c0,1.4-1.1,2.5-2.5,2.5l0,0c-1.4,0-2.5-1.1-2.5-2.5v-5
			C0.9,29.9,2,28.8,3.4,28.8L3.4,28.8z M2,33.8c0,0.8,0.6,1.4,1.4,1.4s1.4-0.6,1.4-1.4c0,0,0,0,0,0c0-0.8-0.6-1.4-1.4-1.4
			C2.6,32.4,2,33,2,33.8C2,33.8,2,33.8,2,33.8z"/>
		<circle id="connector1pin" fill="none" stroke="#9A916C" stroke-width="0.381" cx="39.1" cy="33.8" r="2"/>
		<path fill="#9A916C" d="M10.6,28.8c1.4,0,2.5,1.1,2.5,2.5v5c0,1.4-1.1,2.5-2.5,2.5l0,0c-1.4,0-2.5-1.1-2.5-2.5v-5
			C8.1,29.9,9.2,28.8,10.6,28.8L10.6,28.8z M9.2,33.8c0,0.8,0.6,1.4,1.4,1.4s1.4-0.6,1.4-1.4c0,0,0,0,0,0c0-0.8-0.6-1.4-1.4-1.4
			C9.8,32.4,9.2,33,9.2,33.8C9.2,33.8,9.2,33.8,9.2,33.8z"/>
		<circle id="connector2pin" fill="none" stroke="#9A916C" stroke-width="0.381" cx="46.3" cy="33.8" r="2"/>
		<path fill="#9A916C" d="M17.8,28.8c1.4,0,2.5,1.1,2.5,2.5v5c0,1.4-1.1,2.5-2.5,2.5l0,0c-1.4,0-2.5-1.1-2.5-2.5v-5
			C15.3,29.9,16.4,28.8,17.8,28.8L17.8,28.8z M16.4,33.8c0,0.8,0.6,1.4,1.4,1.4s1.4-0.6,1.4-1.4c0,0,0,0,0,0c0-0.8-0.6-1.4-1.4-1.4
			S16.4,33,16.4,33.8C16.4,33.8,16.4,33.8,16.4,33.8z"/>
		<circle id="connector3pin" fill="none" stroke="#9A916C" stroke-width="0.381" cx="53.5" cy="33.8" r="2"/>
		<path fill="#9A916C" d="M25,28.8c1.4,0,2.5,1.1,2.5,2.5v5c0,1.4-1.1,2.5-2.5,2.5l0,0c-1.4,0-2.5-1.1-2.5-2.5v-5
			C22.5,29.9,23.6,28.8,25,28.8L25,28.8z M23.6,33.8c0,0.8,0.6,1.4,1.4,1.4c0.8,0,1.4-0.6,1.4-1.4s-0.6-1.4-1.4-1.4
			S23.6,33,23.6,33.8L23.6,33.8z"/>
		<circle id="connector4pin" fill="none" stroke="#9A916C" stroke-width="0.381" cx="60.7" cy="33.8" r="2"/>
		<path fill="#9A916C" d="M32.2,28.8c1.4,0,2.5,1.1,2.5,2.5v5c0,1.4-1.1,2.5-2.5,2.5l0,0c-1.4,0-2.5-1.1-2.5-2.5v-5
			C29.7,29.9,30.8,28.8,32.2,28.8L32.2,28.8z M30.8,33.8c0,0.8,0.6,1.4,1.4,1.4c0.8,0,1.4-0.6,1.4-1.4s-0.6-1.4-1.4-1.4
			C31.4,32.4,30.8,33,30.8,33.8L30.8,33.8z"/>
		<circle id="connector5pin" fill="none" stroke="#9A916C" stroke-width="0.381" cx="67.9" cy="33.8" r="2"/>
		<path fill="#9A916C" d="M39.4,28.8c1.4,0,2.5,1.1,2.5,2.5v5c0,1.4-1.1,2.5-2.5,2.5l0,0c-1.4,0-2.5-1.1-2.5-2.5v-5
			C36.9,29.9,38,28.8,39.4,28.8L39.4,28.8z M38,33.8c0,0.8,0.6,1.4,1.4,1.4c0.8,0,1.4-0.6,1.4-1.4s-0.6-1.4-1.4-1.4
			C38.6,32.4,38,33,38,33.8L38,33.8z"/>
		<circle id="connector7pin" fill="none" stroke="#9A916C" stroke-width="0.381" cx="82.3" cy="33.8" r="2"/>
		<path fill="#9A916C" d="M53.8,28.8c1.4,0,2.5,1.1,2.5,2.5v5c0,1.4-1.1,2.5-2.5,2.5l0,0c-1.4,0-2.5-1.1-2.5-2.5v-5
			C51.3,29.9,52.4,28.8,53.8,28.8L53.8,28.8z M52.4,33.8c0,0.8,0.6,1.4,1.4,1.4s1.4-0.6,1.4-1.4s-0.6-1.4-1.4-1.4S52.4,33,52.4,33.8
			L52.4,33.8z"/>
		<circle id="connector6pin" fill="none" stroke="#9A916C" stroke-width="0.381" cx="75.1" cy="33.8" r="2"/>
		<path fill="#9A916C" d="M46.6,28.8c1.4,0,2.5,1.1,2.5,2.5v5c0,1.4-1.1,2.5-2.5,2.5l0,0c-1.4,0-2.5-1.1-2.5-2.5v-5
			C44.1,29.9,45.2,28.8,46.6,28.8L46.6,28.8z M45.2,33.8c0,0.8,0.6,1.4,1.4,1.4s1.4-0.6,1.4-1.4s-0.6-1.4-1.4-1.4S45.2,33,45.2,33.8
			L45.2,33.8z"/>
		<path fill="#9A916C" d="M61.3,28.8c1.4,0,2.5,1.1,2.5,2.5v5c0,1.4-1.1,2.5-2.5,2.5l0,0c-1.4,0-2.5-1.1-2.5-2.5v-5
			C58.8,29.9,59.9,28.8,61.3,28.8L61.3,28.8z M59.9,33.8c0,0.8,0.6,1.4,1.4,1.4c0.8,0,1.4-0.6,1.4-1.4c0,0,0,0,0,0
			c0-0.8-0.6-1.4-1.4-1.4S59.9,33,59.9,33.8C59.9,33.8,59.9,33.8,59.9,33.8z"/>
		<path fill="#9A916C" d="M68.5,28.8c1.4,0,2.5,1.1,2.5,2.5v5c0,1.4-1.1,2.5-2.5,2.5l0,0c-1.4,0-2.5-1.1-2.5-2.5v-5
			C66,29.9,67.1,28.8,68.5,28.8L68.5,28.8z M67.1,33.8c0,0.8,0.6,1.4,1.4,1.4s1.4-0.6,1.4-1.4c0,0,0,0,0,0c0-0.8-0.6-1.4-1.4-1.4
			C67.7,32.4,67.1,33,67.1,33.8C67.1,33.8,67.1,33.8,67.1,33.8z"/>
		<path fill="#9A916C" d="M75.7,28.8c1.4,0,2.5,1.1,2.5,2.5v5c0,1.4-1.1,2.5-2.5,2.5l0,0c-1.4,0-2.5-1.1-2.5-2.5v-5
			C73.2,29.9,74.3,28.8,75.7,28.8L75.7,28.8z M74.3,33.8c0,0.8,0.6,1.4,1.4,1.4s1.4-0.6,1.4-1.4c0,0,0,0,0,0c0-0.8-0.6-1.4-1.4-1.4
			S74.3,33,74.3,33.8C74.3,33.8,74.3,33.8,74.3,33.8z"/>
		<path fill="#9A916C" d="M82.9,28.8c1.4,0,2.5,1.1,2.5,2.5v5c0,1.4-1.1,2.5-2.5,2.5l0,0c-1.4,0-2.5-1.1-2.5-2.5v-5
			C80.4,29.9,81.5,28.8,82.9,28.8L82.9,28.8z M81.5,33.8c0,0.8,0.6,1.4,1.4,1.4c0.8,0,1.4-0.6,1.4-1.4s-0.6-1.4-1.4-1.4
			S81.5,33,81.5,33.8L81.5,33.8z"/>
[... 82 lines cut ...]
		<circle fill="#646464" cx="57.5" cy="12.4" r="1.7"/>
		<circle fill="#646464" cx="87.6" cy="6" r="1.7"/>
		<polygon fill="#646464" points="67.2,4.5 79.6,4.5 80.5,5.3 78.8,7.8 68,7.8 66.4,5.3 		"/>
		<polygon fill="#EF1818" points="81.3,6.2 82.1,7.8 80.5,16.9 78.8,18.6 77.1,16.9 78.8,8.6 		"/>
		<polygon fill="#EF1818" points="78.8,19.4 79.6,21.1 78,30.2 76.3,31.9 74.7,30.2 76.3,21.9 		"/>
		<polygon fill="#646464" points="74.7,33.5 62.2,33.5 61.4,32.7 63,30.2 73.8,30.2 75.5,32.7 		"/>
		<polygon fill="#EF1818" points="60.6,31.9 59.7,30.2 61.4,21.1 63,19.4 64.7,21.1 63,29.4 		"/>
		<polygon fill="#EF1818" points="63,18.6 62.2,16.9 63.9,7.8 65.5,6.2 67.2,7.8 65.5,16.1 		"/>
		<polygon fill="#EF1818" points="63.9,18.6 65.5,16.9 76.3,16.9 78,18.6 76.3,20.3 65.5,20.3 		"/>
		<circle fill="#646464" cx="80.1" cy="31.9" r="1.7"/>
		<polygon fill="#646464" points="96.1,4.5 108.5,4.5 109.3,5.3 107.7,7.8 96.9,7.8 95.2,5.3 		"/>
		<polygon fill="#EF1818" points="110.2,6.2 111,7.8 109.3,16.9 107.7,18.6 106,16.9 107.7,8.6 		"/>
		<polygon fill="#EF1818" points="107.7,19.4 108.5,21.1 106.9,30.2 105.2,31.9 103.5,30.2 105.2,21.9 		"/>
		<polygon fill="#EF1818" points="103.5,33.5 91.1,33.5 90.3,32.7 91.9,30.2 102.7,30.2 104.4,32.7 		"/>
		<polygon fill="#EF1818" points="89.4,31.9 88.6,30.2 90.3,21.1 91.9,19.4 93.6,21.1 91.9,29.4 		"/>
		<polygon fill="#EF1818" points="91.9,18.6 91.1,16.9 92.8,7.8 94.4,6.2 96.1,7.8 94.4,16.1 		"/>
		<polygon fill="#646464" points="92.8,18.6 94.4,16.9 105.2,16.9 106.9,18.6 105.2,20.3 94.4,20.3 		"/>
		<circle fill="#646464" cx="109" cy="31.9" r="1.7"/>
		<polygon fill="#C7C7C7" points="113.4,36.6 -0.1,36.6 0.9,37.9 114.4,37.9 		"/>
		<polygon fill="#AAAAAA" points="113.3,36.6 114.4,37.9 114.4,2 113.3,1 		"/>
	</g>
</g>
</svg>

```

nateevans, 2019-01-23:
@exocode - Know this has been a long time... but after banging my head on this with Illustrator SVGs here's what I found:

The main issue is that in the main `<svg>` tag, there is no `width` or `height` property set. You _could_ manually add those and save the SVG and it should import, but I found the easiest way is to get the SVG Options in Illustrator set correctly.

These settings work for me (set CSS Properties to "Presentation Attributes" and make sure to uncheck "Responsive". After saving like this, I could import to Fritzing no problem.):

[image]

exocode, 2019-01-24:
@nateevans  cool nice trick. I think I can close that issue. Thank you!

closed 2019-01-24 by exocode