fritzing/fritzing-app #3189: parts.db prevent fritzing from loading new parts

el-j, 2016-02-12:
1. add a new part to the fritzing-parts (fzp+all svgs in it's folders)
2. start fritzing
3. fritzing does not find the part

fritzing should check the parts-branch and folders and update the parts.db on load. otherwise the git-update function during fritzing loads needs to pull the new parts and a new parts.db (around 10mb at the moment.) is it possible to "attach" new parts at the end of the parts.db?

aknoerig, 2016-02-14:
Not sure I understand.. 
The parts.db is no longer stored in the git repo. It only gets created by developers when doing a release. 

If you want to experiment with new parts or fix old ones, there is now a new command line option -ov, that points to an override folder. All parts stored there will be loaded after parts.db and extend/override it. 

See also https://github.com/fritzing/fritzing-app/wiki/3.-Command-Line-Options

aknoerig, 2016-02-14:
Wait, now I get it.. 
This should be happening, and I think it works for me. I will check again.

aknoerig, 2016-02-14:
The parts.db is completely regenerated after the user accepts the update, so that seems to work as expected.
 
What exactly did you do that didn't work, step by step?

el-j, 2016-02-14:
ok step by step:

1. make a new part (fzp and svg in the parts-core-folderstructure/fritzing-parts-repository)
2. open fritzing
3. the part cannot be found in fritzing
4. move/delete parts.db
5. open fritzing again
6. fritzing starts with all parts
7. looking for the new part, shows the part via fritzing-parts-search
8. fritzing-parts folder has a new parts.db now with 0kb size
9. close fritzing
10. open fritzing again
[image]
[image]

the only way to solve the problem:
a) copy back parts.db -> fritzing starts but without the new part
b) write a new parts.db -> the complete new parts.db file has to be pushed/pulled (10mb)

the ov option did the job, but slows down the loading and search mechanism. it seams that it works without the parts.db completely. 

the question is now: how can new parts in the parts-repository, be used, without rewriting the hole db-file? otherwise the update-mechanism for users does not make sens from my point of view.

aknoerig, 2016-02-14:
I think you are tricking yourself here, if you try to create a new part in the same fritzing-parts repo that the Fritzing app is using.
Instead, you should keep a separate clone of fritzing-parts (let's name it fritzing-parts-dev) for pushing changes.

So the new workflow for core parts contributors would go like this:
* Test all new parts or changes in the override folder 
* When working, copy these to your fritzing-parts-dev clone. Push from there.
* Run fritzing, which is using the not yet updated fritzing-parts clone. It will then update its clone and regenerate parts.db

el-j, 2016-02-15:
ok, i got it. sorry. 
but when i now click on "update parts" fritzing crashes...

[image]

[image]

next time i reopen fritzing, the update message is gone. it works normal but do not have the new part in the library and the new parts-files aren't cloned into my fritzing-parts-app folder.
[image]

what is wrong now? any idea?

aknoerig, 2016-02-15:
Not really. ;) For me everything is working fine. Please try a clean new setup. If it crashes, the crash report / command line output / Qt Creator error messages would be helpful.

el-j, 2016-02-15:
"http check new version no error" 
is the last message i get from qt. after that, i have to hit: "programm schließen"
and it ends up with "~\Fritzing\debug64\Fritzing.exe exited with code 255"
the windows error log on this:
```
Name der fehlerhaften Anwendung: Fritzing.exe, Version: 0.9.2.0, Zeitstempel: 0x56c1ae91
Name des fehlerhaften Moduls: git2.dll, Version: 0.23.0.0, Zeitstempel: 0x56610937
Ausnahmecode: 0xc0000005
Fehleroffset: 0x000000000018b820
ID des fehlerhaften Prozesses: 0x858
Startzeit der fehlerhaften Anwendung: 0x01d167e1f914f6a8
Pfad der fehlerhaften Anwendung: ~\Fritzing\debug64\Fritzing.exe
Pfad des fehlerhaften Moduls: ~\libgit2\build64\git2.dll
Berichtskennung: f77f0806-e59b-497e-a8c9-a150e39cea7c
Vollständiger Name des fehlerhaften Pakets: 

```

i have found this libgit2 error on stackoverflow:
http://stackoverflow.com/questions/28978460/libgit2-write-tree-object-to-mysql-odb-backend

maybe that helps

closed 2016-03-27 by aknoerig