fritzing/fritzing-app #3193: The part editor is not loading silkscreens

matissev, 2016-02-23:
Hello,

I'm having trouble with the part editor as it is not loading the silkscreen of my part, although the copper layers are fine. It seems that it can recognize the svg path but can't display it black.
However, the schematics works. Didn't try the breadboard view though.
I tried to import the svg file of a standard IC that has a working silkscreen with no success.
I also tried to make a new .fzpz package, to convert units to inches, and to remove all the custom parts in the custom parts folder in order to avoid conflicts, but nothing works.

Here it is.

[pcb.svg.zip](https://github.com/fritzing/fritzing-app/files/142433/pcb.svg.zip)

Thank's in advance.

KjellMorgenstern, 2019-11-20:
This might already be fixed on 0.9.5 develop branch.

DrItanium, 2020-01-04:
Hmmmm, when I try to load the file I get an error with the message:

There is already a part with id '0223_0eb44bbb4c4f50c21567b8c79110a3c3_5' loaded into Fritzing.

Tested with 0.9.5 development branch

KjellMorgenstern, 2020-01-04:
The issue is for sure outdated,
the root cause might or might not be fixed,
the provided A4988 has conflicts with another part which is already obsolete.
A nice A4988 part is available ( Pololu A1)

Therefore I will close the issue.

closed 2020-01-04 by KjellMorgenstern