fritzing/fritzing-app #3195: PCB trace width should remember the last used value

Kazade, 2016-03-01:
If you are laying down your traces in PCB view it's really annoying to after every trace to change the wire thickness. I suggest that the trace thickness should be "sticky" and the next trace you draw should inherit the thickness of the last trace.

DrItanium, 2020-01-04:
I've been designing a ton of boards in the last two weeks and this issue kept coming up. As a addendum to this, there should be a way to denote a default width as well.

cowboydaniel, 2022-06-15:
[image]
[image]
I don't know if this is new or something, but the autorouter trace width IS setable. I just found this setting, and by selecting custom, you can change the default trace width!!!

KjellMorgenstern, 2022-06-15:
@cowboydaniel The issue is not about autorouter trace width, so it is still open. PS. Please avoid cross-posting.

closed 2024-10-10 by KjellMorgenstern