fritzing/fritzing-app #3200: mouse weird actions

vdhub, 2016-03-13:
Greetings, 

I`m using this on El Capitan and it seems that when i move the mouse around the tabs are going like crazy shifting between them (Welcom; breadboard, schematic, pcb, code) and as well on the parts, every time i press on a part inspector control it goes like crazy scrolling in the lists and does no stop until i remove the mouse from that area . I could make a movie if you want with the functionality for you to see. 

Thanks.

aknoerig, 2016-03-14:
Whow, sounds annoying.. That seems to be an issue specific to your setup though. Are you using a graphic tablet (like a Wacom) or something like that?

vdhub, 2016-03-14:
Nope i`m using the touch pad on the mac and the magic mouse , I really cannot use the app at all all the controls goes nuts and if it goes this way to much the app just crashes so i`m searching for an alternative in the meanwhile . or maybe get a virtual machine to try it on windows.

failiz, 2021-02-14:
There is no way to replicate this and the info is not enough to debug. Did you solve the issue? 
Close if no update? (@KjellMorgenstern , I think we should create that label to not forget to close these kind of issues if there is no reply)

KjellMorgenstern, 2021-02-14:
Closing for now, nothing we can do. If you observe the issue with 0.9.4 or later, feel free to reopen or open a new issue.

failiz, 2021-02-14:
@KjellMorgenstern  I think you forgot to close it...

closed 2021-02-14 by KjellMorgenstern