fritzing/fritzing-app #3205: Cannot access any URLs in Fritzing app

daibach142, 2016-04-16:
Version 0.9.2b Linux AMD64 (on Kubuntu Wily (15.10)) fails to access any links - welcome page, FAB, Share and Help items. 'Check for updates' does work - debug log shown here
http://fritzing.org/download/feed/atom/linux-64bit/?version=0.9.2b.11.19.8d2d5970658f0bed09c661c9ea9a515b5f40f44c&sysname=Ubuntu&kernname=Linux&kernversion=4.2.0-35-generic&arch=x86_64&sysversion=15.10&pid=dfba95cda8a15e96c9c91dc7153ee5d8
http check new version
http check new version no error

aknoerig, 2016-04-16:
Weird.. Might have been a temporary server glitch? If not, can you give one concrete, more detailed example?

daibach142, 2016-04-16:
Click any of the help menu items "online tutorials, online projects 
gallery, online parts reference, parts editor help, report a bug, donate 
to Fritzing" all result in nothing visible happening.

Also, clicking "fabricate" or "share" on the PCB tab screen does nothing 
visible.

This has happened every time I've tried it since installing it a few 
days ago.
HTH

Dave Attwood
+44(0)1244 300693
+44(0)7985 625197

aknoerig, 2016-04-20:
I just setup a Ubuntu machine here, but can't verify this. But I assume it has to do with your Kubuntu setup. These links should open in an external browser and it seems that there is none properly defined on your system.  Check in System Settings->Details->Default Applications->Web Browser.
I'm closing this issue since it's not a Fritzing issue.

closed 2016-04-20 by aknoerig

daibach142, 2016-04-20:
I enclose a snapshot of my system setting for the browser, which seems to be honoured by most s/w.
Maybe there is a QT5 bug or something on Kubuntu. I'll get the source code and experiment.
Thanks for your help
[image]

daibach142, 2016-04-20:
Andre, FYI running Fritzing from a shell script shows an error whenever an HTML link is clicked:
`kde-open5: symbol lookup error: /usr/lib/x86_64-linux-gnu/libKF5ItemViews.so.5: undefined symbol: _ZNK19QAbstractProxyModel15canDropMimeDataEPK9QMimeDataN2Qt10DropActionEiiRK11QModelIndex `

landracer, 2016-04-20:
I have personally verified, fritzing works fine in both 15.10 and 16.04 Kubuntu. This does seem to be a system specific issue.

daibach142, 2016-04-20:
Built 0.9.3 from source, and built libgit2. Links work on this version, and now also work on the original failing version 0.9.2b - so it looks like an 'old library' issue.
Thanks to all for your help.

aknoerig, 2016-04-21:
Thanks for investigating this further!
Btw, did you run it via make install, then ./fritzing.sh? I'm getting an error regarding libgit there and had to run it via QtCreator.

daibach142, 2016-04-21:
There was a problem with libgit - I have the following in /usr/lib:
`
/usr/lib/libgit2.so -> libgit2.so.23
/usr/lib/libgit2.so.0.23.4
/usr/lib/libgit2.so.23 -> libgit2.so.0.23.4
`

aknoerig, 2016-04-21:
Thanks! Would you be able to fix the install procedure? I assume either phoenix.pro or fritzing.sh need to be amended, but I'm not sure how.

daibach142, 2016-04-21:
OK will look into it - I expect it will be a slight change on libgit2 installation, rather than to Fritzing. Working ...