fritzing/fritzing-app #3207: Crash whenever a "file dialog" is invoked

davidco206, 2016-05-01:
The same version of Fritzing (from the same download) works fine on one of my older WIn7 machines, but...

Brand new Dell OptiPlex 3020  Win7Pro x64 SP1, Updates applied
Installed Apps: Visual Studio 2013, Office 2010 and Fritzing 0.9.2b11.19 64,  Trend Micro AV
(Anti-virus scans are clean).

When I try "File.Open...",  "File.Save As...", or try to "Import..." a library in the Parts area (or any other command that brings up a file dialog), instead of opening a file dialog as expected, the entire app window greys-out and I get:

       "Fritzing has stopped working  A problem caused the program to stop working correctly. 
        Windows will close the program and notify you if a solution is available. [Debug] [Close Program]"

Clicking [DEBUG] launches VS2013, and displays the following:
"Unhandled exception at 0x000000007777EF57 (ntdll.dll) in Fritzing.exe: 0xC0000005: Access violation writing location 0x0000000000000000."

Do I need a different version of .NET Framework or something?

aknoerig, 2016-05-03:
Whow, that's a weird issue, never seen that before. Fritzing might want to write to its "app local" data folder at that moment, but that should be allowed. Maybe it's the AV software being too prohibitive?

kerry-nz, 2016-09-28:
The problem is with Dell Backup and Recovery. Uninstall it to fix. See http://en.community.dell.com/support-forums/software-os/f/3526/t/19634253 for details.

failiz, 2020-12-29:
Close the issue? It was caused by a third application, we cannot fix this and (hopefully) they already fixed it.

closed 2020-12-29 by KjellMorgenstern