fritzing/fritzing-app #3209: Crash when placing ground fille

noisymime, 2016-05-12:
I am experiencing reproducible crashes when placing ground fills. This occurs anytime the file is saved before an attempt to place the ground fill is made. If the file is simply opened and the ground fill is the first thing to be done, then it works OK. If any modification is made, a save performed and the ground fill placed, the application crashes

Platform is OSX 10.11.4

[schematic v0.4.1.fzz.zip](https://github.com/fritzing/fritzing-app/files/260417/schematic.v0.4.1.fzz.zip)

[Fritzing crash.txt](https://github.com/fritzing/fritzing-app/files/260420/Fritzing.crash.txt)

noisymime, 2016-06-24:
Still getting fill crashes in 0.9.3, however it appears to be a different cause as the crash dump is quite different. Note that the crash dump still gives the version as 0.9.2, however this is definitely running under 0.9.3. 

Can reproduce by placing a ground fill, undoing it, making a change (any change) then trying to redo the ground fill. 

[Fritzing fill crash 2.txt](https://github.com/fritzing/fritzing-app/files/331731/Fritzing.fill.crash.2.txt)

failiz, 2021-02-14:
I could not replicate this in 0.9.4 or 0.9.6rc. Could you try to replicate this in the newest Fritzing version?

KjellMorgenstern, 2021-03-01:
I was able to replicate this in 0.9.6 by removing the sole via on the left, saving, then run ground fill. 
The Fritzing application receives a SIGABRT

KjellMorgenstern, 2023-12-28:
Could not reproduce this with old ground fill algorithm in 1.0.1 , and also not during testing for the 1.0.2 release (old and new copper fill algorithm)

closed 2023-12-28 by KjellMorgenstern