fritzing/fritzing-app #3211: Autrouter ignores single layer setting

chrisspen, 2016-05-16:
I'm trying to design a single-layered board, and even though I've set the PCB's layer to "one layer (single-sided)", when I run the auto-router, it creates a second layer with a ton of vias, which I then have to delete. How do I stop this from happening?

If it's not possible to routing my circuit with a single layer, that's fine, but I'd prefer to get an error message saying "unroutable" instead of it added a bunch of extra parts that I can't use.

YaseenTwati, 2016-06-02:
can you upload your .fzz file ?

YaseenTwati, 2016-07-08:
it works normally here ... maybe a version, os, fzz file or a bit more info ?

failiz, 2021-03-12:
I could not duplicate it in 0.9.6 (Windows) and not enough information

KjellMorgenstern, 2021-03-12:
The autorouter has the ability to add wires to bridge connections when routing is to complex or impossible. I could imagine something similar happens with single sided boards.

failiz, 2021-03-12:
It could be, but it uses jumper for that, not traces on top.
By the way, this looks like a duplicate of #3211

failiz, 2021-03-12:
Sorry, duplicate of #3347

KjellMorgenstern, 2021-03-12:
Closing in favor of #3347

closed 2021-03-12 by KjellMorgenstern