fritzing/fritzing-app #3212: Silkscreen export doesn't include labels

chrisspen, 2016-05-16:
Using any of the "Export->for Production" options produces a silk screen that includes component and pad outlines, but not labels. Is including lables on the silkscreen unsupported, or have I misconfigured something?

For example, given my circuit:

[image]

Fritzing outputs the silkscreen:

[image]

YaseenTwati, 2016-05-31:
are you using a single sided pcb ?
it doesnt usually work for me when exporting in that case , so i just set it it to double sided before exporting 
if it is actually double sided are you sure your looking at the right side ?

chrisspen, 2016-05-31:
Yeah. It's a single sided SMD only board.

YaseenTwati, 2016-06-01:
set the board to double sided and export again 
that tends to work with me

failiz, 2021-03-12:
This is a duplicated of #1942 Yesterday, I could not replicate it because I exported to PDF and then the silkscreen appears. The issue is when you generate for production, then the silkscreen file is blank. Changing the PCB to doubled-side is a workaround (exports well).

KjellMorgenstern, 2021-12-06:
No "mask top" is generated for single sided PCBs. The mask top however was used to keep copper areas clear from silkscreen. Since the mask top was not there, this was handled as "keep all areas clear", resulting in an empty silkscreen-top.

Fix: If there is no "mask top", then we treat all areas as free to use for "silkscreen top".

KjellMorgenstern, 2021-12-06:
Note: A different issue is that SMD components have their labels on silkscreen top. See #3911.

closed 2023-06-23 by KjellMorgenstern