fritzing/fritzing-app #3216: ctrl-s saves a 0KB(corrupt) fzz file

ajvuik, 2016-06-01:
Hello,

I have had this a couple of times now up till the point that I'm done with Fritzing...
What I do is: I press CTRL-S and I see that Fritzing is saving the file. Normaly I check if the file is saved correctly and sometimes it is and sometimes it isn't. But yesterday I forgot to check and lost 5+ hours of work :(
Then when, I try to reload it comes back with an error, because the file is 0KB

I use Version 0.9.2 (b8d2d5970658f0bed09c661c9ea9a5)

aknoerig, 2016-06-02:
That's bad! It's the first time I've heard about this kind of issue so unfortunately there are no good starting points to debug it. Any further clues regarding what you might have done before or during the save? Or maybe some antivirus software getting in the way?

ajvuik, 2016-06-06:
Thanks for your reaction. After cooling down a bit :P I did some tests and it seems to be related with MS Onedrive.
I can recreate the problem while loading and saving directly to Onedrive. Loading and saving from a different location doesn't seem to be an issue. I'll try to write down the steps what I did and see if and how the problem manifests.
As far as I can see it doesn't matter what i do in Fritzing itself.

I also saw that there is a new version of Fritzing. I'll run the update and see if that solves the problem.

aknoerig, 2016-06-07:
Thanks for the further investigation. I updated the ticket description accordingly. Let us know if you fin out more.

bh26, 2020-10-20:
got a similar one.
Saved a file, then tried to open it, and I got an unable to open error message.
I found that using 7Zip to get the .fz file inside allows me to get it anyway.
However I made several tests and it looks like saving it on a synced folder (I use Google drive) creates the problem, while saving it on a local folder works fine.
I put that here in case someone wants to look at it.