fritzing/fritzing-app #3218: Not able to connect to online parts resources

Makoland, 2016-06-03:
I'm not able to connect to the online parts resources for updates due to a proxy server between my PC and the internet.
Is there a way to configure Fritzing to handle a proxy server?

Thanks,
Markus

aknoerig, 2016-06-05:
Whow, good question, we didn't expect a problem with this at all.. There seems to be a recent update to the libgit2 library that handles proxies, but that is not part of a release yet: https://github.com/libgit2/libgit2/pull/3110

If you're a coder, you're welcome to experiment with that. Maybe an update to their master branch already does the trick?

failiz, 2020-10-12:
@Makoland , could you try if this is fixed in 0.9.4, which comes with libgit2 0.28?

KjellMorgenstern, 2020-10-12:
Fritzing doesn't have a dialog to configure a proxy, However it should be possible, with direct access to the parts directory, to configure the got repo there for a proxy.

KjellMorgenstern, 2020-10-12:
So i guess it works, but i don't have a proxy at hand to test. And it needs manual tinkering with the git config of the parts folder.