fritzing/fritzing-app #3219: 3D Functionality

BrainsoftLtd, 2016-06-03:
I'm investigating the possibilities to adapt Eagle3D to work with .fz files instead of Eagle files. I basically want to reuse the POVRAY parts library that ships with Eagle3D and recreate the functionality of the ULP script that ships with it to a script that parses .fz files. The parts library is the main reason for taking Eagle3D as starting point, as I don't know about any 3D part libraries with PCB parts.
Ideally, the dialog used in Eagle for Eagle3D could be included in Fritzing at some point, but to keep dependencies to a minimum I would start by making it an external tool that just uses the .fz files as input.
Since it will be quite an effort, I wanted to check here first if there are other ideas about adding 3D functionality to Fritzing.

aknoerig, 2016-06-05:
Nice! No, there are currently no other plans for this. Some interesting avenues would also be to use Qt's OpenGL capabilities to render the output inside the app. And also take a look at Sparkfun's work on a 3d parts library based on https://github.com/sparkfun/3D_Models.

It's probably much simpler to export the needed information from Fritzing (i.e. write an additional export routine), rather than trying to convert a .fz file. You could take a look at the various exporters that Fritzing has, especially the pick-and-place output.

BrainsoftLtd, 2016-06-24:
Okay, I've been studying the options. Realistic 3D rendering requires two things: raytracing and HDRI lighting. The output is not instant, therefore not interactive, but gives you extremely good quality renderings. POVRAY is the one open source component to achieve these renderings, but it has it's own format and using existing STEP format components (like the Sparkfun's) would require a conversion. The same goes for other open source rendering engines.
Then you proposed looking at OpenGL and in-app rendering. I think this would be a great addition to create an interactive 3D rendering, but from what I have seen about Qt GUI, it seems very involved to get it working compared to most of the rendered scripts, both using native Qt GUI OpenGL and using VTK. Which is a pity, because this feature would be nicest to integrate in the existing app.
The third option, just for the sake of completeness, that is mainly used by Eagle is using SketchUp and the Kerythea renderer. It's easy to create 3D models for components in SketchUp, and there are a lot already available too.
Personally, I only have hands-on experience with POVRAY and the Eagle3D component library, so that will be my starting point. I'll check out the existing exporters to see if I can use that code (although C++ is not my cup of tea), otherwise I'll use the .fz file as a starting point.
I'll use this thread to keep you updated.

BrainsoftLtd, 2016-06-24:
Okay, after rereading that, it didn't feel right... So back to the drawing board, and Google... Resulting in a new proposal. I wanted to revisit the use of STEP files, because of the high availability of components in STEP format, and I wanted to revisit skipping the OpenGL in-app support for the 3D view.
I found another open source project, called Open CASCADE, and it seems to be the missing link. It can read and write STEP files, it supports both standard rendering and raytrace rendering using OpenGL/OpenCL, and is compatible with Qt (even includes tutorials for it). It also offers a high level library to create the geometries by hand (for example for the PCB, traces, holes etc.), which is great, because doing that directly in STEP is close to impossible.
Additional benefit: you will get STEP export for the entire PCB.
I think this is the best of both worlds in terms of interoperability, in-app use of 3D and realistic rendering. If it has your support, I'll base my solution on this.

aknoerig, 2016-06-25:
Yeah, there are quite a number of alternatives when it comes to 3D! Qt also just released Qt3D as part of Qt 5.7, which seems to be a great way to get native 3D inside the app: http://blog.qt.io/blog/2016/06/16/qt-5-7-released/
OpenCASCADE looks amazing, too, but also seems to be a complex beast to get your head around.
Any of these are fine, though I'd first try the minimal approach: create an exporter from Fritzing to STEP, ready to view in any 3rd party viewer. This is what you would need in any case, and will give you a difficult enough challenge to start with. The second "step" could then be to integrate a viewer into Fritzing.

BrainsoftLtd, 2016-06-25:
STEP is indeed the first step, but native STEP is almost impossible to create. OpenCASCADE offers the possibility to import STEP definitions for existing components, and add geometry manually using a nice accesible API (e.g. 'box' to create a box, instead of 50 lines of STEP code). Once you have everything in one model (OpenCASCADE) displaying it in a QT OpenGL window is just as easy as saving the entire PCB as a STEP file for interoperability / external viewers. That's why I said it gives you the best of both worlds. Plus, OpenCASCADE now has a native raytrace renderer compatible with OpenGL, so it fits all the original requirements (export inside the app, display inside the app, and get photorealistic rendering for presentation).

KjellMorgenstern, 2019-11-08:
Closing this issue, as there was no progress and we currently don't have plans to add 3d rendering inside the application. Main issue I see is that for most parts, no 3d information is available.

closed 2019-11-08 by KjellMorgenstern