fritzing/fritzing-app #3220: portuguese translation

Spectrum256, 2016-06-04:
Hi,

I see the new version has been launched, but I see some of the translations I made on the new empty things are not updated, not even in the fritzing credits are updated.

Maybe they have to be updated in the master translation, I mean, in english.

Regards.

aknoerig, 2016-06-05:
Hmm, are they showing in English or in an old Portuguese version? If they are English, then maybe they indeed got reset when the original English was changed once more. I'm afraid you'd then need to re-do these translations. Hope it's not too many?

Spectrum256, 2016-06-06:
Hi,

Yes, I think it was that, because I was comparing the original english version and the portuguese version and it was a reset. I see I lost some updates of the translation.

Well, it was a bit of, but I can make it all over again. But tell me how to do it correctly if I have to update the english one too, to not happen again.

aknoerig, 2016-06-07:
You did everything correctly, the only changes in English besides new items should have been related to some copyright information. So you should be safe from now.

Spectrum256, 2016-06-08:
So I don't need to do anything. I can keep the rest of the translation?

One more thing, can you add me my name in the credits for the global translation? I couldn't did that.

I will keep the translation.

KjellMorgenstern, 2019-05-03:
Changing the global credits in aboutbox.cpp requires updating translations for all other languages. This is quite error prone. Could you file a PR, and we can see if we can improve the process?

closed 2019-05-03 by KjellMorgenstern