fritzing/fritzing-app #3221: Fonts way too big and unconfigurable

gratefulfrog, 2016-06-06:
Since the June 3rd update the fonts are huge and take up too much of the screen.

The tool is nearly unusable.

There seems to be no way to configure the view?

See attached screenshot.

Cheers!
Bob
[image]

aknoerig, 2016-06-07:
Hmm, this seems to be a side-effect of the high-dpi "feature". We didn't have a Linux with retina screen to test, but it works fine on Mac/Win and should be the same on Linux.
What's your monitor setup? Which resolution and how many screens?

We're using the new built-in feature of Qt 5.6 for this, as documented here:
http://blog.qt.io/blog/2016/01/26/high-dpi-support-in-qt-5-6/

And this is how we use it:
https://github.com/fritzing/fritzing-app/blob/master/src/main.cpp#L133

gratefulfrog, 2016-06-07:
I am using a high end Asus laptop UX303LNB with a screen 1920x1080 pixels
at 60Hz.

I am not prepared to start debugging QT...; I would need a setting for font
sizes...

This new version is unusable for me, sigh...

aknoerig, 2016-06-07:
So that's not even retina, weird! There's not going to be an option for font sizes I'm afraid, unless Linux has some system-level support for that?

It's not much debugging to do - it would be great if you could simply try `AA_DisableHighDpiScaling` and see if that works. Then we could put that as an option into Fritzing preferences.

gratefulfrog, 2016-06-07:
sure, but how do I do that?

aknoerig, 2016-06-07:
You'd need to follow the instructions at https://github.com/fritzing/fritzing-app/wiki/1.-Building-Fritzing
It takes a while, but it's not much work, mostly waiting for downloads/compiling.

gratefulfrog, 2016-06-07:
Sorry, but that is way too much work for me, and also too much installing
of software that I don't want...

I'm just going to give up I'm afraid.  I'm pretty disappointed..

nodice73, 2016-06-11:
I can confirm this bug on Linux Mint 17.3. My screen is a 14.1" 1080p Full High Definition ColorPro IPS Display with Matte Surface ( 1920 x 1080 ).

The fonts/graphics are too big. The window menu fonts are the correct size. The window does not allow me to adjust the width, so it starts with some of the window actually outside the frame of my screen.

I reverted to 0.9.2b.

yunkot, 2016-06-14:
I also have this issue and reported it in [this bug report](http://forum.fritzing.org/t/0-9-3b-unusable-on-1080p-display-with-1-25x-scaling/1434).

After reading [this page](http://blog.qt.io/blog/2016/01/26/high-dpi-support-in-qt-5-6), it looks like QT might not really support non-integer display scaling (issues in Fritzing imply that QT applies ceil() on scale factor).

It is possible to revert all this to how Fritzing previously worked by invoking it from command line (from Fritzing folder):
```
QT_AUTO_SCREEN_SCALE_FACTOR=0 ./Fritzing
```

It would be great if such option would be added to Fritzing itself, so on Linux it would start with 1x scale by default and have 2x scale as opt-in feature, at least until QT is ready to support non-2x multi-DPI settings.

nodice73, 2016-06-14:
I just tried running fritzing on my work computer, which is also Linux Mint 17.3 and has two 22" 1920x1080 monitors. In this case, the fonts and scaling looks fine. Does anyone know why?

yunkot, 2016-06-14:
nodice73, you need to check display DPI scaling - your 22'' desktop likely uses 1x, whereas 14.1'' laptop probably uses something like 1.25x, just like mine.

wysiwyng, 2016-07-09:
I can confirm this on Arch Linux with Fritzing installed from AUR. Fonts and everything are way too big, unless that command line switch is passed to Fritzing on starting. Screen is a 14" 1920x1080 Laptop.

gl2748, 2016-07-13:
Same issue. Dell Lattitude E7450
[image]

Xoffis, 2016-07-17:
Hi! my first post, sorry for bad eng!

Now i will stirr in the pott!

I can confirm this bug on win10 too!

[image]

Funny thing is it dont happen every time, i have workt with it in normal size 2 days now, and i restarted fritzing just now and it was super big!

Have happend before...

CoolGames, 2016-07-24:
Here is Ubuntu Studio 16.04 on System 76 Lemur with 1920 x 1080 screen size.
I tried the scale = 0 as well.
All the window controls are off screen beyond the edges but can be found by sliding onto surface.
The min max icons have no effect.
The default window has zoom in out grayed out so not usable either.
The Zip included QT5 5.6.0 but not sure what is being used with QT5 5.5.1 installed by repository or manufacturer along with working 9.2 from repository.

[image]

RebelCoderRU, 2016-10-29:
Same here. Any news about this Bug? Not only scaling is broken, but windows position and resizing too. It is unusable at the moment.

[image]

ovidiub13, 2016-10-30:
This has been posted before, but I can't find it now. This is a workaround:

Set the command in the **Fritzing.desktop** file to:
`QT_AUTO_SCREEN_SCALE_FACTOR=0 /opt/fritzing/Fritzing %u`

Change the path to the Fritzing binary to point to your own.

Right click on **Kickoff** (Application Launcher) -> **Edit applications** -> **Development** -> **Fritzing** -> **Command**

The category under which you'll find it, might be differed. Or you might not have it at all, in which case I recommend you add a new element and give it all the Fritzing details.

tyeth, 2016-11-04:
Change in desktop file caused error launching application for me - ubuntu1604. Running by hand in terminal works fine and resolves the issue. 
Please patch asap.
Also confused to not see maximize/restore buttons on either screen, but they are present with the fix in place.
My setup was laptop (fine) and hdmi tv (huge scale dpi issue).

hansvermeersch, 2016-11-20:
I have the same problem on Windows 8.1. Screen resolution 1920x1080 60Hz. Does anybody know a workaround for windows?
[image]

rkline1963, 2016-11-20:
If you are not on KDE Plasma, then create a bash script with the QT_AUTO_SCREEN_SCALE_FACTOR=0 and the full path to your install of Fritzing for me it is:
`#!/bin/bash
QT_AUTO_SCREEN_SCALE_FACTOR=0 /usr/local/fritzing-0.9.3b.linux.AMD64/Fritzing
`

Make it executable and put that into your path and/or put the path to your file the .desktop file or equivalent for your non-KDE desktop to have a clickable icon.

ovidiub13, 2016-11-20:
The  `.desktop` thing I mentioned is for all Unix desktops. The KDE Plasma this is just how to create it using the GUI.

rkline1963, 2016-11-23:
Putting the

|QT_AUTO_SCREEN_SCALE_FACTOR=0 /opt/fritzing/Fritzing %u|

into the .desktop file for Xfce did not work.  The desktop is expecting 
to see "Exec=filename" in the .desktop file where filename is an 
executable file, and it does not like setting the 
QT_AUTO_SCREEN_SCALE_FACTOR first.

pief, 2017-02-21:
You have to use ``env`` in order to be able to pass environment variables in an ``Exec`` line, like this:

```
Exec=env QT_AUTO_SCREEN_SCALE_FACTOR=0 Fritzing
```

sle44102, 2017-03-05:
I could not get Exec=env QT_AUTO_SCREEN_SCALE_FACTOR=0 Fritzing to work in Linux Mint 18

What did work was this:  Exec=sh -c "QT_SCALE_FACTOR=0.5 /usr/share/fritzing/Fritzing"  in my /usr/share/applications/fritzing.desktop file.

QT_AUTO_SCREEN_SCALE_FACTOR=5 /usr/share/fritzing/Fritzing from the command line also works to make it fit the screen, but the parts Icons were kind of small and hard to read.  This method makes them a little better to read right from the menu.

akkana, 2017-06-27:
This is still an issue in the 0.9.3b that's in the debian stretch repos. On a 1366x768 laptop, the window comes up fullscreened (no titlebar or windowmanager decorations) with only about half the window showing, and HUGE fonts, and no way to change it. QT_AUTO_SCREEN_SCALE_FACTOR=0 fritzing brings up a more reasonably sized window. This is under X11 and openbox.

This auto scale factor stuff is fairly broken if it thinks 1366x768 is a high-DPI screen that needs super high resolution and big fonts.

nopdotcom, 2017-07-22:
It’s better to have a new Fritzing release than to block on this. Please revert to old behavior on Windows and Linux via ifdef at https://github.com/fritzing/fritzing-app/blob/master/src/main.cpp#L133 and call it a day. I don’t think you can get adequate feedback until an alpha or beta is in the field, and you can get at logs.

There’s the separate problem that alphas don’t seem to be buildable on the Mac; see #3339 for a fix.

@akkana, please submit a bug report with Debian if you’re seeing this problem; regardless of upstream, this behavior is clearly broken in their distro, and they may wish to hotfix it. Feel free to cc: nop@nop.com and blame me.

jgoldgh, 2017-07-26:
The font scaling in 9.3b needs work.  I've got a Dell 9560 with the 3840x2160 display running on Linux Mint Cinnamon 18.2. The menu fonts are too small and are basically unreadable.

[image]

You can see the differences in the font sizes.  At my normal working distance from the screen, reading the menu feels like I'm trying to read the last line on the optometrist's eye chart. 

To be fair, lots of programs have trouble running Linux on this monitor.  Pretty much anything Java based has the same issues.

MatejKovacic, 2017-09-05:
I can confirm this issue on Ubuntu Festy (17.04), 64-bit. I installed Fritzing from Ubuntu repositories (with apt install).

However, if I open the console and type:
`QT_AUTO_SCREEN_SCALE_FACTOR=0 Fritzing`
...it runs just nice.

25pinjo15, 2017-09-21:
Have same isues on windows 10. The ui is far too big. Verry not confortable.. No solution at all ? I search in app folder and registery but nothing..

nt314p, 2018-02-27:
Is there anything you can do on windows?

KjellMorgenstern, 2019-11-20:
Anyone with this issue on 0.9.4 yet?

deshipu, 2019-11-20:
I just have:

```
export QT_SCALE_FACTOR=1
export QT_AUTO_SCREEN_SCALE_FACTOR=0
```
in my profile, and that makes all qt apps behave. Without it, they will have randomly sized fonts depending on which display they appear on. I don't think this is a problem with Fritzing itself.

akkana, 2019-11-23:
0.9.4 doesn't seem to be downloadable from fritzing.org yet. I havn't used Fritzing from git before; if I check out the git repo and try to run Fritzing.sh, it just complains that it can't find lib/Fritzing. If I run install_fritzing.sh, it makes a symlink ~/.local/bin/Fritzing that points to ~/.local/share/fritzing/Fritzing which doesn't exist (~/.local/share/fritzing seems to be a copy of the source directory, and so doesn't have a file called just Fritzing in it). Any tips on how to test 0.9.4?

failiz, 2021-11-08:
No new reports since two years. Has this been fixed?

KjellMorgenstern, 2024-03-10:
There were multiple issues causing this.
- Windows / App specific non integer multiplcator zoom  (e.g. 150 %). Fritzing would switch to 200%.
- High resolution displays on linux, especially in multi monitor setups.
- Conflict with wayland configurations

All these have been fixed/mitigated, and the issue has not been reported for a long time.

closed 2024-03-10 by KjellMorgenstern