fritzing/fritzing-app #3223: pins / headers

drawkula, 2016-06-07:
I cannot get "double" row headers/pins... the "double" setting simply seems to be ignored.
(fritzing-0.9.3b-linux-amd64 on Debian8)

aknoerig, 2016-06-07:
Same behaviour on Win & Mac. In PCB view you do get the correct footprint, and pin number is correct in all views. Still confusing, I admit!

philrich123, 2016-07-05:
Pin spacing seems to have the same problem. OK in PCB view but 0.1in in breadboard view whatever choice you do.

Intrinsically-Sublime, 2017-11-22:
Although it is true you can not get double row in Breadboard view you would not likely never do this in real life because it would short out one row to the other and if you are using double row on a PCB you may actually be using the two rows for different connections thus not wanting them shorted in breadboard. Also with pin spacing again you would always want them 0.1 for breadboard because the breadboard is always 0.1 spaced so making them smaller would only make them incompatible. I believe this issue should be closed as it seems to work as intended and would not work if changed.

drawkula, 2017-11-22:
What if you use the prototype PCBs instead of a breadboard?

(I cannot test it right now, the current Fritzing dislikes my Debian9.)

Intrinsically-Sublime, 2017-11-22:
I think in the case of using perfboard in breadboard view you would just use two single row headers since you are basically using it to make pictures not pcbs. And even if you are first prototyping in breadboard view with perfboard you can still put both single row headers side by side in PCB view and it will make holes that will also work with a double header. The only issue with doing so in PCB is the silkscreen will have extra lines but they will be under the part when assembled so again it would not actually cause any issues using it that way. But if it were always double row in Breadboard it would break most everyones sketches prior to now and only help the odd case were someone is using perfboard.

Also when you say the current Fritzing do you mean the Binary download or the Github head?

drawkula, 2017-11-22:
> Also when you say the current Fritzing do you mean the Binary download or the Github head?

The binary download.

Intrinsically-Sublime, 2017-11-22:
Have you reported the issue with Fritzing on Debian9? If not could you create one and add any information you can on the issue. Starting it from a terminal should give you some information and if it will run but has issues you could enable debugging in the help menu.

NerdyProjects, 2018-07-18:
I  just stumbled upon the same error:
I am (trying to) use fritzing to design a perfboard using the breadboard view to hand out to people for whom I am giving a workshop...

sgparry, 2019-02-16:
This is an issue if you are using strip board - you cut the tracks between the pin pairs  - or if you are plugging stuff in the xio of an arduino Mega - or a dozen other use cases.

KjellMorgenstern, 2019-05-03:
Perfboard and strip board views would be feature requests, this is not supported right now.

closed 2019-05-03 by KjellMorgenstern

deshipu, 2019-05-03:
Perfrboard and strip board views very much exist already in Fritzing. You just use the breadboard view, delete the breadboard and drag-and-drop a perfboard from the palette.