fritzing/fritzing-app #3228: 0.9.3b: Cannot use system boost anymore

michaellass, 2016-06-08:
I'm maintaining an unofficial fritzing package for Arch Linux: https://aur.archlinux.org/packages/fritzing/

When trying to package 0.9.3b, build fails with the following message:

> sh: dpkg: Command not found.
> Project MESSAGE: Please download the boost library--you can find it at http://www.boost.org/
> Project MESSAGE: Note that boost 1.54 has a bug in a function that Fritzing uses, so download some other version
> Project ERROR: Copy the boost library to .../src/lib/, so that you have .../src/lib/boost_1_xx_0

Boost is installed on the machine (version 1.60.0) and fritzing 0.9.2b is perfectly happy building against the already installed boost version. Is this an intended change or some regression? I found #2366 which leaves the impression that it should be possible to build against the system boost libraries.

Also I had to manually extract and build libgit2 as described in the wiki although libgit2 is installed on the system (in version 0.24.1 and not 0.23.4 as required according to the wiki, though…). Would it be possible to build against the system library here, too?

All in all I'd like to keep the redundancy of installed libraries as low as possible. Are there any switches I missed?

aknoerig, 2016-06-09:
We'd like to keep it that way, too! 

Regarding boost, it should still take the local version, and at least it does so on my ubuntu machine. We did make some changes to the include logic, maybe that's incompatible with archlinux?
See https://github.com/fritzing/fritzing-app/commit/2a97c2fb5e070dcf4a5bacdca021d58b9d53bdf2

Regarding libgit2, we'd need to add a few ifdefs since the API between 0.23.4 and 0.24.1 slightly changed. Could you give that a try yourself?
All git-related code is in https://github.com/fritzing/fritzing-app/blob/master/src/version/partschecker.cpp, and the include logic is here: https://github.com/fritzing/fritzing-app/blob/master/phoenix.pro#L164

michaellass, 2016-06-09:
Thanks a lot for these pointers! It's pretty clear that the boost include logic currently only works on distributions using dpkg. So also RedHat, Fedora, Suse, Gentoo etc. currently won't use the system boost. I will see if I can find another way to determine the version of boost installed, so no access to a package manager is required. As a first step I now know what to patch to allow building on Arch Linux.

I'll also have a look into the libgit2 API changes.

cjmayo, 2016-06-12:
I've created #3233 with the patch that I have been use to get 0.9.3b to compile with libgit2-0.24.0. But I haven't used any of the git functionality from within Fritzing.

At the moment I just patch out all the boost and libgit2 code:
[fritzing-0.9.3b-boost.txt](https://github.com/fritzing/fritzing-app/files/310830/fritzing-0.9.3b-boost.txt)
[fritzing-0.9.3b-libgit2.txt](https://github.com/fritzing/fritzing-app/files/310831/fritzing-0.9.3b-libgit2.txt)

michaellass, 2016-06-13:
My quick&dirty patches to create an Arch Linux package look quite similar: https://aur.archlinux.org/cgit/aur.git/tree/?h=fritzing

I tried to update a user-writable parts collection using the built-in update functionality but it didn't work, so I'm not sure if git_remote_connect is happy getting NULL as parameter. But I have to double check that against a Ubuntu built that should work.

To circumvent the package manager to check the boost version one could try to look into the corresponding header file:
`grep "define BOOST_LIB_VERSION" /usr/include/boost/version.hpp | cut -d'"' -f2`
The issue is that the location of this header file has to be known. I guess we cannot just import the corresponding libraries (boost and libgit2) to evaluate such version constants?

cjmayo, 2016-06-13:
The example
https://libgit2.github.com/libgit2/ex/v0.24.0/network/ls-remote.html#git_remote_connect-4
has
```
error = git_remote_connect(remote, GIT_DIRECTION_FETCH, &callbacks, NULL);
```

cjmayo, 2016-07-03:
I've had a go here: https://github.com/fritzing/fritzing-app/pull/3248

Still skips Boost detection if installing for system version, but some ideas listed.

aknoerig, 2016-07-04:
#3248 is now merged - @michaellass plz give it a try and maybe add your boost detection as an alternative ifdef?

michaellass, 2016-07-04:
#3248 looks great. It solved the libgit2 issue for me. Regarding boost, please see my pull request which should work on a broader range of systems. But it still has to be tested on a Debian-based system.

aknoerig, 2016-07-05:
#3251 is now merged - does this solve this issue?

michaellass, 2016-07-05:
#3251 works a treat. Thanks a lot!

closed 2016-07-05 by michaellass