fritzing/fritzing-app #3229: 0.9.3b crash at launch complains about Git2.dll

kc2uez, 2016-06-10:
Hi, I get an error when I start fritzing on a old (Circa 2009, has an LPT port) but still powerful computer i7/3ghz/16gb/120gb SSD.
 
It runs XP (I know, but it needs XP because it runs an old special software to program some unique machine via LPT, doesn't work using virtual machine or newer versions of windows). Os is 32 bits with latest updates.

I downloaded Fritzing 0.9.3.b, unziped it and every time I run it, it opens in a few seconds and then this window pops up: http://imgur.com/OfW65lH

If I click on Ignore just bring the box back. If I click on abort it crashes and closes. If I click on retry it  crashes and ask if I want to send a report to MS.

aknoerig, 2016-06-11:
Hmm, it seems like libgit2 only supports Windows Vista and up: https://github.com/libgit2/libgit2/pull/3596
libgit2 is required for the new auto-update feature of the parts library. 
I'm afraid this means that Fritzing no longer supports Windows XP. :( Unless someone creates a workaround, you will have to stay with 0.9.2.

kc2uez, 2016-06-30:
I found by accident a crazy workaround. If I open one instance of fritzing and leave it open with the error message and launch a second instance of fritzing it will works just fine without any issues. Maybe useful for other diehards running xp wanting to use fritzing.

closed 2019-05-04 by KjellMorgenstern