fritzing/fritzing-app #3235: Mystery part's wire connection display broken after flip or rotation.

ZeroNullBit, 2016-06-13:
Mystery part's wire connection display broken after flip or rotation.

Bug re-trig steps:
1. Drap 3 Mystery part into Schema view and place them.
2. Right click the U2 part, Click "Flip Horizontal" menu item.
3. Right click the U3 part, Click "Rotate 180°".
4. Connect the pins each other.
5. Drag the parts' label "U1","U2","U3" to another position.
6. Save the file and close Fritzing.
7. Launch Fritzing again and reload the file just saved.
8. Switch to Schema View , you'll find that the wires of "U2","U3" part are display broken.

ZeroNullBit, 2016-06-13:
OS: Windows 7 64Bit Chinese
Fritzing: 0.93b 64bit.

KjellMorgenstern, 2020-02-04:
Verified the issue
Before:
[image]

After:
[image]

It seems the mystery parts body is not displayed correctly when loaded from a file.

KjellMorgenstern, 2021-12-09:
Might be related to #3180

closed 2023-07-14 by KjellMorgenstern