fritzing/fritzing-app #3244: OBS Build fails with missing libgit2 while this is installed

mvdw, 2016-06-28:
See:
https://build.opensuse.org/package/show/home:m_vanderwulp:branches:Education/fritzing

I am trying to update the fritzing openSUSE RPM package to the latest version.
But building fails due to lack of libgit2, despite it being installed.

This is an extract from the build log:

[   21s] [166/309] keeping libgit2-21-0.21.0-2.1.7
...
[   30s] [267/309] keeping libgit2-devel-0.21.0-2.1.7
...
[   40s] + cd fritzing-app-0.9.3b
[   40s] + qmake-qt5
[   40s] Project MESSAGE: Fritzing requires libgit2
[   40s] Project MESSAGE: Build it from the repo at https://github.com/libgit2
[   40s] Project MESSAGE: See https://github.com/fritzing/fritzing-app/wiki for details.
[   40s] Project ERROR: libgit2 include path not found in ../libgit2/include
[   40s] error: Bad exit status from /var/tmp/rpm-tmp.O1BaB1 (%build)

I have no clue what is wrong here... help please!

cjmayo, 2016-07-03:
See also https://github.com/fritzing/fritzing-app/issues/3228 (it's about libgit2 as well as boost).

Perhaps you could test (and comment on!) PR https://github.com/fritzing/fritzing-app/pull/3248

I think it will need some patches from master to apply:
https://github.com/fritzing/fritzing-app/commit/a795b9f388e74d0a4043a06a9107b664db6605d2
https://github.com/fritzing/fritzing-app/commit/620f85dbc31f376303839ed24cf4c6b0f55f3859
https://github.com/fritzing/fritzing-app/commit/9bcffaf26301a4bb6161a0c712ca481629cffce8
(The first one is only for an error on install_fritzing.sh with the second one)

aknoerig, 2016-07-04:
Merged PR #3248 -- please give it a try on openSUSE, too!

closed 2016-07-04 by aknoerig