fritzing/fritzing-app #3245: Doesn't work with Fedora 23 out of the box (libssl issue)

maweki, 2016-06-29:
I tried running Fritzing on Fedora 23 and I get the following error message:

`/home/maweki/bin/fritzing-0.9.3b.linux.AMD64/lib/Fritzing: error while loading shared libraries: libssl.so.1.0.0: cannot open shared object file: No such file or directory`

I have openssl-devel installed in version `1.0.2h`. So I tried to symlink that lib to `libssl.so.1.0.0` (the same with libcrypto) and I get the following:

`/home/maweki/bin/fritzing-0.9.3b.linux.AMD64/lib/Fritzing: /lib64/libssl.so.1.0.0: version `OPENSSL_1.0.0' not found (required by /home/maweki/bin/fritzing-0.9.3b.linux.AMD64/lib/libgit2.so.23)`

After deleting the distributed libgit2.so-file, Fritzing started. Why is Fritzing distributed with a libgit that is linked with some very specific libssl-version?

Hoall, 2016-07-06:
Can confirm that bug. Same under Fedora 24

maweki, 2016-07-06:
@Hoall remove the `libgit2.so` from the lib-folder that is distributed with fritzing and make sure you have `libgit2-devel` and `openssl-devel` installed on your system.

AlanMcKay, 2017-01-16:
Same problem on CentOS Linux release 7.2.1511 and the libgit2.so fix above did not fix it.  I renamed that file to libgit2.so-renamed

jrobeson, 2017-05-09:
ah, and the suggested solution no longer works on Fedora 25, as the libgit2 version on the system is 0.24 while fritzing expects 0.23.  Fritzing does start, but cannot update the parts library.

maweki, 2017-05-09:
So as I see it, they should distribute a correct libgit2 as well as a libssl or dynamically link to libgit2 0.24. I don't believe that the latter is a problem

zhymer96, 2017-07-13:
/home/vurthe/fritzing/lib/Fritzing: error while loading shared libraries: libssl.so.1.0.0: cannot open shared object file: No such file or directory
i have Parrot OS :c

emnik, 2017-09-19:
We are in Fedora 26 and the problem still remains... :-(

maweki, 2017-09-19:
I am at a conference right now but I removed the libgit from the lib directory and put in my own (from /usr/lib64) and renamed it to `libgit2.so.23`.

This works insofar that the program starts loading and crashes with some `qt.network.ssl` errors. So I would think (and I don't have the time right now) that this should work if one replaces `libQt5Network.so` with the system one. And do this again with the next error.

maweki, 2017-09-19:
Okay, so here's a fuck-it solution that seems to work:
delete all `.so`-files from the lib directory `rm lib/*.so*` and then copy your current `libgit2` into the lib directory `cp /usr/lib64/libgit2.so.2* lib/libgit2.so.23`

emnik, 2017-09-20:
@maweki 
Yes indeed! this works just fine :-)

T0T4R4, 2019-03-25:
Worked. What a brainfuck.

KjellMorgenstern, 2019-08-08:
Fritzing was not build on fedora, and libssl version are not compatible from version to version, so is libgit2, which is not packaged e.g. on ubuntu, therefore fritzing needs to supply it (otherwise enduser would have to build their own version of libgit2).

Anyhow, I am trying to fix this by adding a fedora target to the CI/CD system (travis-ci). Currently I am stuck, locally it builds fine, but on the CD there is a strange error I could not yet track down.

For the error messages, see build
https://travis-ci.org/fritzing/fritzing-app/jobs/568022110
See branch 
https://github.com/fritzing/fritzing-app/tree/fedora

If you want to help, contact me (building and uploading the docker image is not part of the CI, but you could change the docker image name on your branch to try your own fedora images).

I am closing this issue, since fedora 23 is quite outdated, and the suggested solutions (messing with the .so files) can not be recommended. Only proper solution at the time would have been to build fritzing by yourself from source. This is already much easier now by following the information in travis.yml and docker/Docker.fedora files.

closed 2019-08-08 by KjellMorgenstern