fritzing/fritzing-app #3246: White/Grey Stripes on PDF PCB export

YaseenTwati, 2016-07-02:
white stripes over the copper filled parts of a pcb 

[image]

I dont really need to explain just look at the picture ( open it ), this happens with almost all of my boards whenever i export a pdf. Since im home etching i can just open gimp and paint them one by one using the bucket tool, but its an just an extra annoying step when dealing with a complex double sided PCBs
there has been older discussions about it before 

http://fritzing.org/forum/thread/1510/
http://fritzing.org/forum/thread/3936/

but it looks like it hasn't been fixed

aknoerig, 2016-07-04:
Sorry, but don't worry about these! They are just rendering artefacts and won't affect your print or production. The copper fill is created with a scanline algorithm, so it's basically hundreds of black rectangles stacked vertically. There is no gap between them, though, but the PDF viewer sometimes displays them as 1px white/grey lines.

YaseenTwati, 2016-07-06:
Well yes they don't effect the circuit, but they do affect my etch, not sure if that has to do with me choosing the highest quality when printing but yeah as i said they still do. anyways to prove what Im saying I grabbed a camera and took a picture of a print  lol  :D

[image]

aknoerig, 2016-07-07:
Thanks for the photo! Hmm, they are indeed visible.. We'll look into that again, but it's not a high priority since we haven't heard of any issues with it. These lines are too thin for the etching process to have any effect.

rmisko11, 2019-12-26:
They sometimes affect the continuity of my ground plane. 😡  I thought I saw somewhere that someone had a fix a few years ago?

KjellMorgenstern, 2019-12-26:
@rmisko11 Do you have a chance, eg. a VM or a second computer, to try this with the development version of Fritzing (0.9.5 develop, available at github releases for Linux and Mac)?
Also, if you have a simple project that reproduces the issue, it would be really helpful to attach it to this issue.

rmisko11, 2019-12-27:
I have just 0.9.4 and the problem still exists as shown in attached image.
[image]

rmisko11, 2019-12-27:
Kjell,
Thank you for your interest.  I just tried 0.9.5 with the same result. I am a simple project as you requested where you should be able to replicate the bug by simply exporting pdf.  I am using Ubuntu 18.04 LTS.
[SimpleProject.zip](https://github.com/fritzing/fritzing-app/files/4005193/SimpleProject.zip)

KjellMorgenstern, 2019-12-28:
An improvement which on the side fixed this issue was integrated with PR #3203 (going back to #3083) but had to be reverted.

aknoerig, 2019-12-30:
Fwiw, note that the panelizer that was originally blocking #3083 is no longer used and likely won't be used again. (It was part of Fritzing Fab before Aisler)

rmisko11, 2020-03-21:
Is there a plan to fix this soon?

KjellMorgenstern, 2020-03-21:
We have not forgotten about this. Integrating the new export brings a lot of improvements (not only fixing those stripes) and will certainly be done.

rmisko11, 2020-03-21:
Thanks for the rapid response.

Would you be able to give some indication of when the new export might be
implemented?

KjellMorgenstern, 2020-03-21:
Well, let me guestimate :-)
Hacking sessions to integrate it: 1 weekend
Testing and fixing other things that might have broken: 4-8 weeks
Roll out the release: 2 weeks

But before I can start, there is the following
- New stack for the main server
- 0.9.5 release

With that timeline, we end up at end of July as the earliest date for official support, but on the development branch it would already be in May. All quite optimistic estimates.

rmisko11, 2020-03-21:
Great Kjell, I look forward to it.  Thanks.

closed 2024-01-08 by KjellMorgenstern

rmisko11, 2024-01-08:
I just downloaded and tried version 1.02 and the horizontal stripes are
still present in the PDF.

Am I to assume that since #3246 is marked completed that there is no
intention to fix the horizontal stripe problem in PDF?

KjellMorgenstern, 2024-01-08:
@rmisko11 How did you generate the copper fill, did you use the old or the new algorithm?

rmisko11, 2024-01-08:
Thank you for responding so quickly, I think I used the old one, I will try
it again now and get back to you.

rmisko11, 2024-01-08:
Thank you very much, I just did it with the new ground fill and it seems to
work perfectly.

Thanks again,
  Roger