fritzing/fritzing-app #3259: Wrong resolution?

hmrodrigues, 2016-07-13:
When I try to run the app or is too big (on screen 1920x1080) or too small (on screen 1366x768). Either way, I can't resize the window.

I tried using the package manager install and the linux64 from the download page.

**System**
- Arch Linux
- Gnome 3.20.2

**1366x768**
[image]

**1920x1080**
[image]

alan-martinovic, 2016-07-28:
Am also having this issue on Ubuntu 16.04.

Not sure what you mean by `I turn off the 1080p display`...
Physically powering it on and off?

alan-martinovic, 2016-07-28:
Here is a workaround that worked on Ubuntu:

```QT_AUTO_SCREEN_SCALE_FACTOR=0.75 ./Fritzing```

[Reference](https://wiki.archlinux.org/index.php/HiDPI#Qt_5)

jogi-k, 2016-08-05:
I am having the same issue on Ubuntu 14.04 on a Full-HD-Display.
The mentioned workaround from alan-martinovic worked also here.

Jogi

nodice73, 2016-08-10:
I think this is the same issue as #3221. See if you agree.

jws121295, 2016-12-26:
It appears this is still an issue. I don't believe #3221 is related. As mentioned above, setting QT_AUTO_SCREEN_SCALE_FACETOR does the trick. I had to set it to something other than 1. It will work with anything in the form "0.x", however entering different values doesn't seem to have any effect.

TeaPackCZ, 2016-12-31:
I'm afraid, that it has smt to do with this:
From https://wiki.archlinux.org/index.php/HiDPI#Qt_5 :
"Note if you manually set the screen factor, it's important to set QT_AUTO_SCREEN_SCALE_FACTOR=0 otherwise some applications which explicitly force high DPI enabling get scaled twice"

pjockey, 2017-11-22:
Is there a place to put the QT_AUTO_SCREEN_SCALE_FACTOR=0 in order to start the program from the GUI?

pylover, 2018-05-28:
```bash
QT_SCALE_FACTOR=3 QT_AUTO_SCREEN_SCALE_FACTOR=0 Fritzing
```

Fixed my problem on Surface Pro 4(4K screen):
[image]

failiz, 2021-11-08:
Duplicate of #3221
No reports since 2018, fixed?

KjellMorgenstern, 2023-05-25:
DPI settings in Fritzing, Qt and most of the OS / Window Managers have changed a lot. Usually reports do not contain enough information to reproduce similar issues. I observed the issues when using multiple screens that have different DPIs, especially when one of the screens was switched on or off. Often, DPI or zoom settings from the system play an important role.

To mitigate the issue, it would make sense to review Fritzing with the latest recommendations from Qt:
https://doc.qt.io/qt-6/highdpi.html
In general, we will try to let UI elements size more flexible.

But I don't think there is a single bug or missing improvement that would allow for a 'fix'.

closed 2023-05-25 by KjellMorgenstern