fritzing/fritzing-app #3260: Autoroute dialog position with multiple displays.

Deggle, 2016-07-16:
On a multi monitor setup the autoroute dialog centres itself across all monitors, asnd in my case it's awkwardly split between screens. It's a very minor bug, but hopefully an equally simple fix - just a tweak to centre the dialog on current window dimensions rather than total display dimensions.

failiz, 2022-08-08:
I cannot replicate this in 0.9.10 (Windows), see:
[image]

The position is as desired in the code:
AutorouteProgressDialog progress(tr("Autorouting Progress..."), true, true, true, true, pcbSketchWidget, this);
	progress.setModal(true);
	progress.show();
	QRect pr = progress.frameGeometry();
	QRect wr = this->frameGeometry();
	progress.move(wr.right() - pr.width(), pr.top());

KjellMorgenstern, 2022-08-10:
Using a multi-monitor setup on Linux, I have neither observed this.

closed 2022-08-10 by KjellMorgenstern