fritzing/fritzing-app #3261: PNG export of breadboard view shows pixel artifacts across bottom of image

chuank, 2016-07-17:
Purely cosmetic, but still something that impacts the exporting of clean Fritzing images.

When exporting breadboard views in PNG format, there is sometimes one or more horizontal rows of additional pixels at the bottom of the image that's not part of the original view. It looks like leftover pixels from an image buffer. This does not seem to impact the JPG exporter.

Here are some examples:
[image]

[image]

I'm using the OSX version of the application.

EdJoPaTo, 2016-07-26:
got the same on the right side of the exported image (but not on the bottom)

on Arch Linux installed via AUR, v0.9.3

thomasw-mitutoyo-ctl, 2020-08-28:
> got the same on the right side of the exported image (but not on the bottom)
I have a single line of transparent pixels at the bottom.
0.9.4 on Windows 10

KjellMorgenstern, 2020-10-18:
Breadboard fz
[wirec3.zip](https://github.com/fritzing/fritzing-app/files/5397966/wirec3.zip)

Exprot before #3725
[image]

Export after #3725
[image]

So, artifacts at the right are gone.
However there is still a transparent line at the bottom that could look like reported by @chuank , depending on where the image is used.

It is a bit difficult to show those transparent lines, I created a codepen: https://codepen.io/team/fritzing/pen/MWeejyP 
 Note the gap between the images.

failiz, 2020-10-19:
Strange, I open those images in Gimp (size is 366x282) and I could not find the transparency. The alpha channel was set to 255 in all the pixels. Could it be an artifact of the viewer?

thomasw-mitutoyo-ctl, 2020-10-20:
Other machine here, does not produce a transparent bottom line. Version 0.9.4  (CD-498-0-a1ffcea 2019-12-01) 64 [Qt 5.12.3] Windows 10. Could this be a library/DLL issue?

KjellMorgenstern, 2020-10-20:
My mistake, the transparent line is not there.

PR #3725 (2020-10-20): Fill image with the background before exporting, fixes #3261
fixes #3261
commit 72c8be18 (2020-10-20): Fill image with the background before exporting, fixes #3261
closed 2020-10-20 by KjellMorgenstern