fritzing/fritzing-app #3262: Geber export svg stroke-width bug

ThomDietrich, 2016-07-23:
While defining my own part I stumbled over a nasty bug in the way svgs are converted which only effects the gerber export function and therefor may stay undiscovered before manufacture.

My part got translated to a nice set of lines on silkscreen in pdf or svg export (first image), so everything seemed fine. However the gerber export will produce thick unwanted lines (second image).

[image][image]

The following change in `pcb.svg` was needed to fix that problem for my part: https://github.com/fritzing/fritzing-parts/pull/73/files

ThomDietrich, 2016-07-23:
Unsure if this is related to and will be solved by https://github.com/fritzing/fritzing-app/pull/3203
@Kazade would you be able to test the part?

el-j, 2017-03-31:
thanks for point this out. 
it is known that every connector needs his own stroke-width element to export properly. 
the stroke-width element on cooper-group on any other group is not allowed.

this is more a parts issue than a fritzing app-issue. 
the [fritzing-parts](https://github.com/fritzing/fritzing-parts) repository needs automatic checks for such issues. it is planed to do these checks + corrections for every contributed part in future.

close this because of parts-issue

closed 2017-03-31 by el-j

ThomDietrich, 2017-03-31:
Hey @el-j, closing the ticket in favor of an referencing parts issue would have been fine by me. Without the referencing parts issue you lost this ticket, are you sure that's how you want to deal with this?

el-j, 2017-03-31:
ok, you are right.
so maybe we can have the issue like that:
https://github.com/fritzing/fritzing-parts/issues/94

closed again ;)

ThomDietrich, 2017-03-31:
Thanks ;) Keep up the great work