fritzing/fritzing-app #3263: Box selection too greedy

deshipu, 2016-07-31:
Hello, I really love Fritzing, and I create most of my small projects with it. It's the most convenient and usable PCB design program I tried. Thank you for your work. However, there is one annoyance in the user interface that has been bugging me for years now, and I decided to write about it here, hoping to start a discussion and maybe eventually implement an option that would fix it for me and others who feel the same.

When you make a rectangular selection, Fritzing will select all the parts that touch the rectangle you've drawn. That is very different from other programs, such as Inkscape, which would only select the objects that entirely fit inside the rectangle. This small quirk makes it very hard to work with smaller parts sitting on top of a larger part -- such that a component on top of a PCB -- because every time you use the rectangular selection, the larger part gets selected, and you might even not notice it if working at larger zoom levels. Then you zoom out, and see that your whole design is ruined, because when you moved the small part, the large part moved too. Or you sit there and stare at the grayed out rotate icon, wondering why you can't rotate the parts you selected. Or a number of similar issues.

I would like to bring into consideration creating an option of making the rectangular selection less greedy -- that is, to make it only select elements that fit inside the rectangle by default. But perhaps such an option already exists, and I just failed to find it?

deshipu, 2021-06-02:
Apparently many applications make the selection "greedy" (selecting any element that even partially touches it) when the mouse is dragged from top to bottom, but non-greedy (only parts entirely inside the selection are selected) when dragged from bottom to top. This is a bit difficult to discover, but would provide both behaviors.

KjellMorgenstern, 2021-06-03:
There already are more or less obvious solutions/workarounds:
a. By <kbd>Ctrl</kbd> + <kbd>Left Mouse</kbd> you can select and unselect parts from a selection
b. You can make parts sticky: <kbd> Right Mouse</kbd> on the part, then toggle 'Sticky'

deshipu, 2021-06-03:
For sure there are ways, but they don't always work. Especially the ctrl+left mouse is frustrating when the bigger part is always preferred (and the send to back/bring to front commands don't seem to do anything). I usually end up having to move the larger part away, noting its original coordinates, move the smaller part, and then type the coordinates for the larger part back.

Also note how making the selection non-greedy by default would solve the problem with selecting the PCB every time you select anything — most of the time the PCB wouldn't fit whole inside the selection, except for the one case where you want to select it and select everything.