fritzing/fritzing-app #3264: Fritzing crash on open/save using gnome and wayland backend

razerraz, 2016-08-06:
When using gnome wayland session with GDK_BACKEND set to wayland, fritzing crash on opening open/save dialogs (gtk dialogs) with the message :
gdkwindow-x11.c:5555 drawable is not a native X11 window

The fritzing process is loaded into a Xwayland session (ie QT_QPA_PLATFORM not set to wayland-egl)

Forcing GDK_BACKEND to X11 solve the issue

It should be expected to have Fritzing handling correctly Xwayland as X server without forcing the backend.

fritzing 0.9.3b-2
wayland 1.11.0-1
gtk3 3.20.8-1

whitelynx, 2019-07-28:
Is this still an issue for you?

KjellMorgenstern, 2019-11-08:
We have a build of Fritzing for Fedora since some time now, and no complaints. To my knowledge, Fedora uses Wayland by default, so I think this is solved.

closed 2019-11-08 by KjellMorgenstern