fritzing/fritzing-app #3267: Windows (64-bit) download - Flagged by Windows Defender as containing a Virus

jbdoucet, 2016-08-13:
Windows Defender flags the 'fritzing.0.9.3b.64.pc' download as containing a virus (Trojan:Win32/Varpes.M!cl):

_

> Category: Trojan
> 
> Description: This program is dangerous and executes commands from an attacker.
> 
> Recommended action: Remove this software immediately.
> 
> Items: 
> containerfile:C:\Users\xxx\Downloads\fritzing.0.9.3b.64.pc.zip
> file:C:\Users\xxx\Downloads\fritzing.0.9.3b.64.pc.zip->fritzing.0.9.3b.64.pc/Qt5Core.dll
> file:C:\Users\xxx\Downloads\fritzing.0.9.3b.64.pc.zip->fritzing.0.9.3b.64.pc/Qt5SerialPort.dll

_

reischle, 2016-08-15:
Same here. Fresh Win10 Install with all parches, but without any additional Antivirus software apart from defender.

jezdean, 2016-08-16:
Windows Defender flags fritzing.0.9.3b.64.pc with the same virus.

Nothing is found if I upload the files to Metadefender. 

* [Qt5SerialPort.dll(https://www.metadefender.com/#!/results/file/cf0db084848546fc97859f50cfc830f2/regular/analysis)
* [Qt5Core.dll](https://www.metadefender.com/#!/results/file/881c34bb77b14dc0a0e62e80210ddb0d/regular/analysis)

jezdean, 2016-08-16:
Windows Defender flags fritzing.0.9.3b.64.pc with the same virus.

Nothing is found if I upload the files to Metadefender. 

* [Qt5SerialPort.dll](https://www.metadefender.com/#!/results/file/cf0db084848546fc97859f50cfc830f2/regular/analysis)
* [Qt5Core.dll](https://www.metadefender.com/#!/results/file/881c34bb77b14dc0a0e62e80210ddb0d/regular/analysis)

aknoerig, 2016-08-17:
Thanks, everyone, that's really weird.. Shouldn't be anything wrong with our files. The Qt framework we're using is absolutely trustworthy, so it's probably a glitch in Defender. Is there a way to report it as a false positive?

gieemek, 2016-11-24:
Norton Antivirus flags as virus:
Qt5Svg.dll
in frizing.0.9.3b.64.pc and remove it. Ofcourse Fritzing can't start. So something is wrong with this framework.

BaraPellumb, 2020-07-22:
Why is this closed?!! should only be closed once a solution is found. Anyhow i say this as   windows defender flag should not just be ignored Why? because i got this other flag from my trusty AVG antivirus:  here: 
[image]
Could some copies of frizzing have be compromised ? Please do not just close it!

KjellMorgenstern, 2020-07-23:
@BaraPellumb It is closed because the issue was identified as a false positive for this 4 year old version of Fritzing.

Which version of Fritzing are you using?
From the error message, it looks like a false positive again. More recent versions of Fritzing check if it has write access before installing its database. Many users tried to copy Fritzing to directories which are write protected, or only accessible by admins. The file test.txt is part of a check to prevent this. Fritzing should be allowed to create and delete this file, otherwise it will likely also run into issues with saving and updating its database.

closed 2020-07-23 by KjellMorgenstern

BaraPellumb, 2020-07-24:
Today I bought  a new copy  from your  website . I think you should  put
more security  on it but my problem  may be  now  be solved.  To me looks
odd  and I'm not entirely  convinced by your  email.  I do think that some
copies circulating  on Internet are bring tampered  with hence the flags.
You should not allow fake copies  to be updated  as that may comprehensive
your servers  and then  spread through updates. I like fritzing  but you
may need to allow people  who pay a donation to  download  new copy  should
they need update or if they change computer.  Also fake copies  Must not be
allowed to have updates via your  servers.
Thanks