fritzing/fritzing-app #3268: Fritzing Build fails in QT (Win 7, 64bit), Error : libgit2 library not found in ../libgit2/build64

ak1801, 2016-08-16:
Hi,

I am trying to build fritzing project in Qt. I have followed the steps described in this blog - https://github.com/fritzing/fritzing-app/wiki/1.-Building-Fritzing1.

I built libgit2 with Visual Studio compiler using the following commands & the libraries are compiled under libgit2/build64

..\fritzing\libgit2\build64>cmake -G "Visual Studio 14 2015 Win64" ..
..\fritzing\libgit2\build64>cmake --build .

Now, when I am trying to run the project in Qt, getting the following error:

01:40:53: Running steps for project phoenix...
01:40:53: Starting: "C:\Qt\5.6\msvc2015_64\bin\qmake.exe" C:\Users\aksmahaj\Documents\MyLab\fritzing\fritzing-app\phoenix.pro -r -spec win32-msvc2015 "CONFIG+=debug" "CONFIG+=qml_debug"
Info: creating cache file C:\Users\aksmahaj\Documents\MyLab\fritzing\build-phoenix-Desktop_Qt_5_6_1_MSVC2015_64bit-Debug\.qmake.cache
Project MESSAGE: target arch: x86_64
Project MESSAGE: found libgit2 include path at C:/Users/aksmahaj/Documents/MyLab/fritzing/fritzing-app/../libgit2/include
Project ERROR: libgit2 library not found in ../libgit2/build64
01:40:53: The process "C:\Qt\5.6\msvc2015_64\bin\qmake.exe" exited with code 3.
Error while building/deploying project phoenix (kit: Desktop Qt 5.6.1 MSVC2015 64bit)
When executing step "qmake"
01:40:53: Elapsed time: 00:00.

It is able to locate libgit2 include path but not able to find libgit2 library. I believe that my setup is correct & the library is generated in git2.dir.

The error is related to the following code snippet in phoenix.pro :
I believe the control is going in else block.
[image]

packagesExist(libgit2) {
    PKGCONFIG += libgit2
} else {
    include(pri/libgit2detect.pri)
}

Please help me fix this issue. I have attached screenshot of my libgit2/build64 directory for reference.

aknoerig, 2016-08-17:
Hi! Note that our wiki instructions state that you need to place the compiled libgit library into a 'libgit2/build64' folder (which you might need to create manually).

ak1801, 2016-08-18:
Hi, 

I have compiled libgit library into 'libgit2/build64' folder as you can see in the screenshot in my previous post. 

As directed in the wiki, I followed these steps sequentially:

1. downloaded & unzipped libgit2-0.23.4.zip in same directory as fritzing-app (as sibling)
2. renamed to libgit2
3. C:\Users\aksmahaj\Documents\MyLab\fritzing>cd libgit2
4. C:\Users\aksmahaj\Documents\MyLab\fritzing\libgit2>mkdir build64
5. C:\Users\aksmahaj\Documents\MyLab\fritzing\libgit2>cd build64
6. C:\Users\aksmahaj\Documents\MyLab\fritzing\libgit2\build64>cmake -G "Visual Studio 14 Win64" ..
    output:
-- The C compiler identification is MSVC 19.0.24210.0
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio 14.0/VC/bin/x86_amd64/cl.exe
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio 14.0/VC/bin/x86_amd64/cl.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Could NOT find PkgConfig (missing:  PKG_CONFIG_EXECUTABLE)
-- Could NOT find ZLIB (missing:  ZLIB_LIBRARY ZLIB_INCLUDE_DIR)
-- zlib was not found; using bundled 3rd-party sources.
-- LIBSSH2 not found. Set CMAKE_PREFIX_PATH if it is installed outside of the default search path.
-- Looking for futimens
-- Looking for futimens - not found
-- Looking for qsort_r
-- Looking for qsort_r - not found
-- Looking for qsort_s
-- Looking for qsort_s - found
-- Looking for clock_gettime in rt
-- Looking for clock_gettime in rt - not found
-- Found PythonInterp: C:/csvn/Python25/python.exe (found version "2.7.1")
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/aksmahaj/Documents/MyLab/fritzing/libgit2/build64

7. C:\Users\aksmahaj\Documents\MyLab\fritzing\libgit2\build64>cmake --build .
(this builds the project without any errors).

Do you think I have missed anything?

ak1801, 2016-08-26:
Hi André,

I debugged this issue and found the root cause, the issue was with the path. 

Environment:
I was facing lots of troubles with win64 bit, so I did the setup the environment in win32 bit.
using Qt32 bit, vs 14 32 bit.

I found 2 ways of resolving it. 

1. By changing LIBGIT2LIB path variable in libgit2detect.pri

When I build libgit2 (libgit2-0.23.4), it builds "git2.lib" under "../libgit2/build32/Debug" folder by default. But the libgit2detect.pri file in fritzing is expecting this library to be present at "../libgit2/build32" folder.. see the following code from libgit2detect.pri:

`win32 {
    contains(QMAKE_TARGET.arch, x86_64) {
        LIBGIT2LIB = ../libgit2/build64
    } else {
        LIBGIT2LIB = ../libgit2/build32
    }

    exists($$LIBGIT2LIB/git2.lib) {
        message("found libgit2 library in $$LIBGIT2LIB")
    } else {
        error("libgit2 library not found in $$LIBGIT2LIB")
    }
} 
`
The hack I did below:

`win32 {
    contains(QMAKE_TARGET.arch, x86_64) {
        LIBGIT2LIB = ../libgit2/build64
    } else {
        LIBGIT2LIB = **../libgit2/build32/Debug**
    }
    exists($$LIBGIT2LIB/git2.lib) {
        message("found libgit2 library in $$LIBGIT2LIB")
    } else {
        **message**("libgit2 library not found in $$LIBGIT2LIB")
    }
}
`
2. Second way is instead of changing the path, I manually copied all the files from "../libgit2/build32/Debug" to  "../libgit2/build32/" and recompiled fritizing in Qt.

Clearly, the issue was with loading of the library from the path fritzing is expecting. Also, I had to manually copy git2.dll in the build directory for running fritzing.exe.

Hope my analysis helps someone else!

Thank you.

ak1801, 2016-09-04:
Hi Andre,

I am following up with you on this bug. Do you have any thoughts on my previous analysis?

oyeluckydps, 2017-06-17:
Hello ak1801,

I am facing the same trouble as you. I tried the hack posted by you but it doesn't seem to work for me.The error message that I am getting in the 'Issues' section of QT Creator is

:-1: error: libgit2 library not found in ../libgit2/build/Debug

I used the default Visual Studio and did not specify the version or architecture while compiling libgit. I have changed the name of folder from 'libgit2.x' to 'libgit2' after compilation but that shouldn't be a problem.

[image]
This is the screenshot of the Debug folder.

I couldn't follow you on the instruction in your last message
"Environment:
I was facing lots of troubles with win64 bit, so I did the setup in win32 bit (using Qt32 bit, vs 14 32 bit)."
Kindly explain this again.

Thanking You
Abhinav

KjellMorgenstern, 2019-05-15:
libgit2 now adds a release or debug subdirectory on windows  because windows build systems are ignoring some of the build settings. So the folder might be ../libgit2/build64/release or ../libgit2/build64/debug now, depending on your libgit2 build. The updated release.bat script (currently on develop branch) takes this into account and expects  ../libgit2/build64/release.

closed 2019-05-15 by KjellMorgenstern