fritzing/fritzing-app #3270: Search field is not working

igdr, 2016-08-25:
Hi,

I installed a new version (Version 0.9.3), but search field doesn't work at all.
[http://prntscr.com/ca83ov](url)

OS: Windows 10 x64

Thanks.

SteveSzyd, 2016-09-04:
Hi
I have identical problem with Fritzing 0.9.3b.
OS: Windows 7 32 bit.

bcelary, 2016-09-16:
OS: Windows 7 / 64-bit. Exact same problem.

Skipperro, 2016-09-21:
OS: Linux Mint 17.3 64-bit. After installing 0.9.2 version there is no search function at all.

umarsear, 2016-11-09:
Same problem, Windows 10 64bit. 

This renders the application practically unusable!

umarsear, 2016-11-09:
Downgraded to 0.9.1b and the search works.

Fergo, 2017-01-03:
Same issue with Fritzing 0.9.3 BETA, running on Windows 10 64bits.

beneuto, 2017-03-02:
Same problem.   Can't use Fritzing effectively without Search.  Can't use earlier version as I need the newer parts.

vanepp, 2017-03-02:
You folks may be interested in the discussion on this issue going on in the fritzing forums at 

http://forum.fritzing.org/t/search-function-in-the-parts-view-does-not-work/1536/27

I so far can't recreate the problem on either Win7 or Win10, but believe the problem lies in the user directories that Fritzing creates, although deleting them has been reported to not work for at least 
someone with this problem. A post in the older forums has also been just found that indicates the same thing. Instructions on moving aside the directories is in the forum posts (be careful because the directories contain all your parts and sketches!).

beneuto, 2017-03-06:
This was resolved for me by upgrading to Windows 10 Anniversary Update.

lagunacomputer, 2017-06-11:
i tried everything.  no search box to type in.  win 7 64 and .9.3b.  i should note i have fritzing running from a second hard drive M:\.

umarsear, 2017-06-12:
The only thing that worked for me was to revert to a previous version, can't remember which.

vanepp, 2017-06-12:
If you don't see the search box field at all, scroll down in the parts bins. At least once (and for one person that we tried everything else before remembering to suggest this) I have seen the search bin at the bottom of the list of bins rather than the top as is usual. If it is just below the scroll window you can drag it up to the top where it normally is.

Braehead, 2018-08-13:
I too had the same problem on Windows 10, with the search icon not appearing in the parts bin from a fresh extraction.  I have only started to use Fritzing for the first time in the last few days and was really stuck without this feature, so....

This morning, I tracked it down to my anti-virus software (in my case BitDefender) silently blocking writes to:
c:\users\adam\onedrive\documents\fritzing\bins

I found a message in the event log for the AV, so 'allowed' / granted the program access to write to these sorts of protected areas including the AppData area for profiles and all is functioning.  I have a search button - yeah.

Also, due to the updating of a fresh installation, I have found it best NOT to have Fritzing installed on a network share, it takes _forever_ to start up the first time and killing the process before this update completes it is not good either.

Hope this helps others and may explain why it has been difficult for some to pinpoint.

Janus303, 2021-03-09:
There is no search field visible in 0.9.6 on MacOS Catalina for me.

KjellMorgenstern, 2021-03-09:
@Janus303 Do you have a file "search.fzb" in $HOME/Documents/Fritzing/bins/search.fzb  ?

$HOME could also be iCloud, so it would show as
"iCloud Drive > Documents > Fritzing > bins > search.fzb" in Finder. The file should have
the fritzing icon, with the letters "fzb" on it.

What happens if you open that file in Fritzing? Right click or two finger click -> open .
It should open Fritzing and show the last search result.

Any additional information might be useful. Some things to check:

What is the file size of search.fzb ?

If you rename the file, and then start fritzing, you should have two search bins
one at the top, one at the bottom of the bin list.

If you delete the file, Fritzing should recreate it once you start it,
and it should be about 15kB size.

Do you have write access to the Documents/Fritzing/bin directory,
is it on a removable media? Or is it synced via iCloud with other
machines that run older versions of Fritzing?

KjellMorgenstern, 2021-03-13:
I think it is most likely that Fritzing is prevented from writing the file.
That happen due to antivirus or network drives.
In any case, it might help to show an error message when the file can not be created, showing the path to the user. That way it should be much easier to identify the various causes.

zzjlamb, 2021-12-27:
Fritzing has no search field on my Mac (Big Sur). In my machine the Fritzing folder in $HOME/Documents is an alias to the Fritzing package folder in Applications. I replaced the alias with a plain empty folder called Fritzing, started Fritzing, and the search field now works.

Reynolds-Beta, 2022-05-10:
Similar issues with Windows 11. Installed 0.9.9 and there's no search function at all.

KjellMorgenstern, 2022-05-10:
@Reynolds-Beta Do you have a file "search.fzb" in $HOME/Documents/Fritzing/bins/search.fzb ?

Reynolds-Beta, 2022-05-10:
Created the folder manually, started Fritzing and now search shows up. (folder path I created: **%userprofile%/Documents/Fritzing/bins**). It is at the bottom of the bins list, but that's easy to rectify. So, it seems there's an issue with Fritzing that doesn't allow it to create a folder, even under the user's context. Not entirely sure why that is, as other programs (eg. Arduino IDE) have no problems creating folders in the user's Documents folder.

This could be fixed in the installer, and core feature installation like this should be done at the install level versus the runtime level. At the very least, add an installer step to create the folder structure.

KjellMorgenstern, 2022-05-10:
It is not application data, but user data. It stores search results, but also custom (imported) parts, edited parts and the like. The installer could run with admin permissions, which would make this issue much worse.

Reynolds-Beta, 2022-05-10:
The core issue seems to be that Fritzing (the app) silently fails creating this folder structure at **%userprofile%/Documents/Fritzing/bins**. No debug output. As this is a core feature, it would be best that the folder structure and feature gets installed at installation time versus at runtime. Alternatively, the installer could make the folder structure so that if it fails an error can be added to the setup logs.

KjellMorgenstern, 2022-05-10:
Do you have any antivirus or the like installed, or are you using a unsual configuration (e.g. network filesystem)? Not sure what would prevent Fritzing from creating a directory with user permissions.

Reynolds-Beta, 2022-05-10:
> If you mean %userprofile% instead of %HOME% (Windows doesn't have %HOME%) then no, there's no Fritzing subfolder there. That's not where the installer puts Fritzting.
> 
> Fritzing gets installed to: %userprofile%/AppData/Local/Programs/Fritzing
> 
> Parts bins are located here: %userprofile%/AppData/Local/Programs/Fritzing/fritzing-parts/bins
> 
> There is no search.fzb in this or any subfolder
> 
> In fact, searching the whole hard drive, there doesn't appear to be a file search.fzb anywhere.

This is the location Fritzing gets installed to when you choose "Only this user" (versus "All users") at installation time.

Reynolds-Beta, 2022-05-10:
> 

Other apps (such as Arduino IDE) have zero problems installing folders under %userprofile%/Documents, so I doubt this is an antivirus issue. Windows Defender didn't flag anything as unusual when Fritzing starts.

Reynolds-Beta, 2022-05-10:
> Do you have any antivirus or the like installed, or are you using a unsual configuration (e.g. network filesystem)? Not sure what would prevent Fritzing from creating a directory with user permissions.

Windows Defender didn't flag anything as unusual when Fritzing starts, so I doubt this is an antivirus blocking problem. Other apps have no problem creating folders in %userprofile%/Documents - Arduino IDE, for example.

closed 2026-10-26 by KjellMorgenstern