fritzing/fritzing-app #3271: Any files can not be open with 0.9.3b

moicci, 2016-08-26:
Hi,

I updated Fritzing version 0.9.3b and open fritzing file written by 0.9.3 or 0.9.2.
But it shows a message "Unable to open 'FILENAME': out.open(): %s".

debugging log:
`file open /Users/fukuda/eval/GPIO.fzz
Setting platform to Arduino
Setting board to Arduino UNO
Setting board to Arduino UNO
Setting port to cu.Bluetooth-Incoming-Port
Setting platform to Arduino
unzipping 11 entries from /Users/fukuda/eval/GPIO.fzz
out.open(): %s`
``

All of my fritzing files could not be open with the error.

Then I tried to execute "Part > Regenerate parts database" because the files are includes my own parts, but it reports an error "Unable to open temporary file".

Let me know how to solve this issue.

Thank you in advance.

gwheng, 2016-10-15:
i have the same issue and this all happen on version "fritzing.2013.07.27.pc" & "fritzing.0.8.7b.pc" in windows 10 Pro x64 . i using a laptop with win7 Pro open all the file without the error. In win7 Pro virtual machine have same issue on all the version...

gwheng, 2016-10-15:
i manage to open the file in windows 10 Pro x64 by rename the file name without Chinese character and not under any subfolder with Chinese character. hope this hint help others...

KjellMorgenstern, 2019-07-18:
It seems opening files with chinese charactes in the filename or path caused a problem in Version 0.9.3. Can someone confirm this is fixed or still a problem with the development version available via github releases?

closed 2019-07-18 by KjellMorgenstern