fritzing/fritzing-app #3279: Check for updates

donaghan, 2016-09-25:
Unable to  install updates. (sorry unable to download new parts)
[image]

Outdoorsman, 2016-11-03:
I installed version 0.9.3b, the Windows 64 bit version and am getting this exact same thing. To reproduce do this...

1. Select "Help > Check for updates..."
2. Click "Update parts" from the popup (pic 1)
3. Get the error notice popup (pic 2)

[image]

[image]

KjellMorgenstern, 2019-07-19:
The error message must be improved to give more details about the problem (configuration error, write error, network error...)

KjellMorgenstern, 2021-07-13:
The error message was improved and a check for writing permissions was added. For now (0.9.7) the feature is disabled. It still caused issues (eg. when the connetion to github is slow). Also it is much easier for us to release new versions of Fritzing and Fritzing parts at once. 
Updating parts via git can still be done manually.

closed 2021-07-13 by KjellMorgenstern