fritzing/fritzing-app #3281: Cursor still appearing on unselected notes

Choumat, 2016-09-27:
After editing and unselecting a note, the cursor remains visible inside it

bartgrefte, 2018-09-20:
The cursor is also visible after exporting to png and pdf, not on the exported svg.

oasis9, 2021-01-21:
Can confirm this is still an issue

failiz, 2021-02-07:
OK. I tested it in 0.9.6rc (Windows) and the error is still there but this bug does not always appear. 
Steps to reproduce:
1. Open Fritzing.
2. Go to BB view
3. Add a note
4. Try to change the text
The cursor remains when you unselect the note

Interestingly, this behaviours does not happen in the SCH view. And a couple of times I could not reproduce it in the BB view after placing a note in the SCH view. 
Add bug label.

failiz, 2021-02-07:
OK, after a bit more testing I found a couple of things.
If you open Fritzing and the you go to file->new and then try to reproduce the bug in the second instance, th bug does not show up.
It also does not appear if you open Fritzing and a file at the same time (by double clicking a file or by passing the path of the file as argument). It seems a bit weird, initialization issue?

failiz, 2021-04-10:
In addition, the left arrow and right arrow move the note, instead of moving the cursor.

failiz, 2021-07-29:
Difficult to debug as sometimes is not easy to reproduce (but it happens very frequently). 
Until now, I found that only happens in the BB view. Not sure why this happens. In other views, the NoteGraphicsTextItem::focusInEvent and NoteGraphicsTextItem::focusOutEvent are fired as soon as you select the text inside the note (or unselect it). This does not always happens in the notes in the BB. I could not also reproduce it if you open a file in Fritzing or open a new app. Thus, a workaround is to save the file and open it again. 
Any ideas about what could cause this?

KjellMorgenstern, 2024-10-22:
Testing with Fritzing 1.0.4: This might have changed since the initial bug report was filed.

There are three different ways a note could be selected:
[image]
[image]\
[image]

The selection of the text inside the note is not affected by the menu option "Edit -> Deselect". This happens always, and should be fixed. 
I could not see any random behavior as described in this ticket.