fritzing/fritzing-app #3282: a problem pcb lm2596

atrinsoft, 2016-10-19:
hello
there is a problem in lm2596 module design.
in pcb part the legs are very near to each other that is encounter a problem and not correct.
the distance between legs should be standard. sincerely yours
[image]
[image]

vanepp, 2016-10-20:
Yep, sure looks like it. While I hate to say it, you would be better raising this in the Fritzing forums. This repository is mostly dead (hopefully it won't stay that way, but as of now with no commits since July it sure looks that way). I happen to have one of these modules and a pair of vernier calipers (and more amazingly could find the module!) so I can fix this up and drop it in parts submit in the forums. At present the forums are where all the activity is occurring. 

Peter Van Epp

vanepp, 2016-10-21:
It actually has more problems than that, breadboard view is quite broken as well. However as noted since I have a couple of these I had an interest so I fixed (maybe anyway) the 2 bugs. The fixed part is available on the Fritzing forums at 

http://forum.fritzing.org/t/correction-for-the-riorand-lm2596-from-core-somewhere/2272

hope it helps, if you cut a board I'd appreciate knowing if the holes actually line up properly :-) 

Peter Van Epp