fritzing/fritzing-app #3285: Problems with the download.

Sever44, 2016-11-11:
Please help

[image]

vanepp, 2016-11-14:
While I've not seen this error, I'd check that the platform folder contains qwindows.dll (which it does on my working Win7 system) as that seems to be what it is looking for without success. If its not there on your system as it suggests try a reload of Fritzing (and/or check that that file is in the download that you are using). If it is in the zip file but not on your system check your antivirus program isn't deciding its a virus.

np511, 2017-04-05:
You need to extract the program fist - you're trying to run it through 7-Zip.

vanepp, 2017-04-05:
Good catch! I completely missed that that was a 7zip window ...

Peter

np511, 2017-05-23:
This can be closed right? No need to distract people with a solved issue...

closed 2019-05-03 by KjellMorgenstern