fritzing/fritzing-app #3286: Making life difficult for package maintainers

hsanjuan, 2016-11-20:
I have tried to fix OpenSUSE packages to provide Fritzing 0.9.3b and it has only come with problems when building. Latest version available was 0.9.2b and now it:

* assumes libgit2 is downloaded inside the project and cannot use system-provided one
* libgit2 is doing an awful job at keeping a stable API. It is not clear which libgit2 version Fritzing should be built against. i.e. 0.24 or 0.22 APIs change and Fritzing needs patching in both cases.
* assumes we are on Ubuntu when checking for system boost libraries

Those things can be patched, ok. When starting Fritzing it makes a lot of assumptions about the location of the parts. It does not check a regular place for parts like `/usr/share/fritzing`. Even when passing `-f /usr/share/fritzing` there are error messages because it expects `/usr/share/fritzing/bins` to exist instead of `/usr/share/frintzing/parts/bins` which would be the logical as the `bins` is distributed in the fritzing-parts repo.

In short, latest upgrades have surely broken about every package build out there and made it difficult to maintain the package. Ideally, Fritzing should:

  * Allow to deactivate libgit dependency. Parts are installed in the system and not controlled by the user, new parts are distributed via a new package so this feature is not usable in systems where Fritzing has been installed with a package manager.
  * Do not do libboost guess work magic. Just let the user compile against whatever is installed in the system, if something is.
  * Check `/usr/share/fritzing` as folder containing parts, sketches, translations by default. Bins are inside `parts/bin`. At least, have a build flag for this or a mechanism.

hsanjuan, 2016-11-23:
Upon strace-ing, I discovered `Fritzing` looks at `/usr/share/fritzing-parts` for parts which is nice, and must be getting translations/sketches from `/usr/share/fritzing` because it doesn't error anymore. So my 3rd proposal can be crossed out (although documentation about this would help).

heinervdm, 2017-02-04:
@hsanjuan I'm the openSUSE Fritzing maintainer, and you got all the points why I haven't updated to the latest release yet. But now I' got it to work at least for Leap 42.2 and Tumbleweed, so that I will publish it soon.
Still the automatic update of the parts library from the git repository will not work yet...

hsanjuan, 2017-02-04:
@heinervdm that's great! I've just realized that I have deleted my Education repository branch with the necessary patching to the Fritzing sources :( I hope it wasn't too difficult to get through the problems. I can have a look if you want and do some testing.

heinervdm, 2017-02-04:
I had most stuff done shortly after the relaese, but due to lack of time some parts were missing, these I've fixed now. Now I'm only fighting with boost, to get the correct version ;-)

If you have a half working update, you can also open a bug report at the opensuse bugtracker and point me to your work. Then I will have a look at it, help is always welcome.

hsanjuan, 2017-02-04:
@heinervdm I remember fixing  boost by patching https://github.com/fritzing/fritzing-app/blob/master/pri/utils.pri

but there have been a couple of commits since I touched that file and now it uses qtCompileTest rather than `dpkg` (which i guess it's good). The bad part is that OpenSUSE comes with boost 1.54 which is apparently buggy. I just removed the safeguards and used it and I find any relevant problem... I don't know how critical it is the buggy function in that version.

heinervdm, 2017-02-04:
In the beginning I've worked around this by just removing these tests, now Ive included the patches from @cjmayo.
My problem with boost was that I wanted a version different from 1.54, but this is now also fixed. I've just commited Fritzing 0.9.3b to Factory.

KjellMorgenstern, 2019-05-03:
There are multiple issues collected here, some of which have been solved.
Fritzing builds with 0.24 to 0.28 libgit2
For boost, the following ticket was created: https://github.com/fritzing/fritzing-app/issues/3453

KjellMorgenstern, 2019-05-03:
For the part folder: https://github.com/fritzing/fritzing-app/issues/3454

KjellMorgenstern, 2019-05-03:
Split issue into two new issues, closing this one.

closed 2019-05-03 by KjellMorgenstern

gasinvein, 2019-12-18:
Since 0.9.4 most downstream patches could be dropped. Thanks for your effort, @KjellMorgenstern!