fritzing/fritzing-app #3289: Trouble with libssl on Korora 24 (Fedora 24 derivative)

mostlybob, 2016-11-25:
This is similar to #3245, but for Fedora 24, it seems.

I have `libgit2-devel` and `openssl-devel` installed.

Here are the steps I've taken:

Out of the box, in the extracted folder, I run `./Fritzing` and get:

``` shell
~/Downloads/fritzing-0.9.3b.linux.AMD64/ ./Fritzing
/home/bob/Downloads/fritzing-0.9.3b.linux.AMD64/lib/Fritzing: error while loading shared libraries: libssl.so.1.0.0: cannot open shared object file: No such file or directory
```

Running `ldd ./lib/Fritizing` shows that `libssl.so.1.0.0` and `libcrypto.so.1.0.0` are both not found:

``` shell
~/Downloads/fritzing-0.9.3b.linux.AMD64/ ldd ./lib/Fritzing 
	linux-vdso.so.1 (0x00007fffa83f2000)
	libz.so.1 => /lib64/libz.so.1 (0x00007fa96790e000)
	libgit2.so.23 => /home/bob/Downloads/fritzing-0.9.3b.linux.AMD64/./lib/libgit2.so.23 (0x00007fa9675e8000)
	libQt5Svg.so.5 => /home/bob/Downloads/fritzing-0.9.3b.linux.AMD64/./lib/libQt5Svg.so.5 (0x00007fa967394000)
	libQt5PrintSupport.so.5 => /home/bob/Downloads/fritzing-0.9.3b.linux.AMD64/./lib/libQt5PrintSupport.so.5 (0x00007fa967127000)
	libQt5Widgets.so.5 => /home/bob/Downloads/fritzing-0.9.3b.linux.AMD64/./lib/libQt5Widgets.so.5 (0x00007fa9668b4000)
	libQt5Gui.so.5 => /home/bob/Downloads/fritzing-0.9.3b.linux.AMD64/./lib/libQt5Gui.so.5 (0x00007fa9660bd000)
	libQt5Xml.so.5 => /home/bob/Downloads/fritzing-0.9.3b.linux.AMD64/./lib/libQt5Xml.so.5 (0x00007fa965e82000)
	libQt5Network.so.5 => /home/bob/Downloads/fritzing-0.9.3b.linux.AMD64/./lib/libQt5Network.so.5 (0x00007fa965b22000)
	libQt5Sql.so.5 => /home/bob/Downloads/fritzing-0.9.3b.linux.AMD64/./lib/libQt5Sql.so.5 (0x00007fa9658de000)
	libQt5SerialPort.so.5 => /home/bob/Downloads/fritzing-0.9.3b.linux.AMD64/./lib/libQt5SerialPort.so.5 (0x00007fa9656c5000)
	libQt5Core.so.5 => /home/bob/Downloads/fritzing-0.9.3b.linux.AMD64/./lib/libQt5Core.so.5 (0x00007fa964fb2000)
	libstdc++.so.6 => /lib64/libstdc++.so.6 (0x00007fa964c2a000)
	libm.so.6 => /lib64/libm.so.6 (0x00007fa964921000)
	libgcc_s.so.1 => /lib64/libgcc_s.so.1 (0x00007fa964709000)
	libc.so.6 => /lib64/libc.so.6 (0x00007fa964347000)
	libssl.so.1.0.0 => not found
	libcrypto.so.1.0.0 => not found
	librt.so.1 => /lib64/librt.so.1 (0x00007fa96413e000)
	libpthread.so.0 => /lib64/libpthread.so.0 (0x00007fa963f22000)
	libGL.so.1 => /lib64/libGL.so.1 (0x00007fa963cae000)
	libgobject-2.0.so.0 => /lib64/libgobject-2.0.so.0 (0x00007fa963a5c000)
	libgthread-2.0.so.0 => /lib64/libgthread-2.0.so.0 (0x00007fa96385a000)
	libglib-2.0.so.0 => /lib64/libglib-2.0.so.0 (0x00007fa96354b000)
	libXext.so.6 => /lib64/libXext.so.6 (0x00007fa963339000)
	libX11.so.6 => /lib64/libX11.so.6 (0x00007fa962ff9000)
	libicui18n.so.56 => /home/bob/Downloads/fritzing-0.9.3b.linux.AMD64/./lib/libicui18n.so.56 (0x00007fa962b5e000)
	libicuuc.so.56 => /home/bob/Downloads/fritzing-0.9.3b.linux.AMD64/./lib/libicuuc.so.56 (0x00007fa9627a6000)
	libicudata.so.56 => /home/bob/Downloads/fritzing-0.9.3b.linux.AMD64/./lib/libicudata.so.56 (0x00007fa960dc2000)
	libdl.so.2 => /lib64/libdl.so.2 (0x00007fa960bbe000)
	/lib64/ld-linux-x86-64.so.2 (0x000055726c407000)
	libexpat.so.1 => /lib64/libexpat.so.1 (0x00007fa960991000)
	libxcb-dri3.so.0 => /lib64/libxcb-dri3.so.0 (0x00007fa96078e000)
	libxcb-present.so.0 => /lib64/libxcb-present.so.0 (0x00007fa96058b000)
	libxcb-randr.so.0 => /lib64/libxcb-randr.so.0 (0x00007fa96037c000)
	libxcb-xfixes.so.0 => /lib64/libxcb-xfixes.so.0 (0x00007fa960174000)
	libxcb-render.so.0 => /lib64/libxcb-render.so.0 (0x00007fa95ff6a000)
	libxcb-shape.so.0 => /lib64/libxcb-shape.so.0 (0x00007fa95fd65000)
	libxcb-sync.so.1 => /lib64/libxcb-sync.so.1 (0x00007fa95fb5e000)
	libxshmfence.so.1 => /lib64/libxshmfence.so.1 (0x00007fa95f95b000)
	libglapi.so.0 => /lib64/libglapi.so.0 (0x00007fa95f72b000)
	libselinux.so.1 => /lib64/libselinux.so.1 (0x00007fa95f504000)
	libXdamage.so.1 => /lib64/libXdamage.so.1 (0x00007fa95f301000)
	libXfixes.so.3 => /lib64/libXfixes.so.3 (0x00007fa95f0fa000)
	libX11-xcb.so.1 => /lib64/libX11-xcb.so.1 (0x00007fa95eef8000)
	libxcb-glx.so.0 => /lib64/libxcb-glx.so.0 (0x00007fa95ecdf000)
	libxcb-dri2.so.0 => /lib64/libxcb-dri2.so.0 (0x00007fa95ead9000)
	libxcb.so.1 => /lib64/libxcb.so.1 (0x00007fa95e8b7000)
	libXxf86vm.so.1 => /lib64/libXxf86vm.so.1 (0x00007fa95e6b1000)
	libdrm.so.2 => /lib64/libdrm.so.2 (0x00007fa95e4a1000)
	libpcre.so.1 => /lib64/libpcre.so.1 (0x00007fa95e22e000)
	libffi.so.6 => /lib64/libffi.so.6 (0x00007fa95e026000)
	libXau.so.6 => /lib64/libXau.so.6 (0x00007fa95de21000)
```

The files are available, though:

``` shell
~/Downloads/fritzing-0.9.3b.linux.AMD64/ ls -al /lib64/libssl*.so*   
-rwxr-xr-x. 1 root root 314456 Nov  2 08:29 /lib64/libssl3.so
lrwxrwxrwx. 1 root root     16 Sep 26 06:25 /lib64/libssl.so -> libssl.so.1.0.2j
lrwxrwxrwx. 1 root root     16 Sep 26 06:25 /lib64/libssl.so.10 -> libssl.so.1.0.2j
-rwxr-xr-x. 1 root root 467696 Sep 26 06:25 /lib64/libssl.so.1.0.2j

~/Downloads/fritzing-0.9.3b.linux.AMD64/ ls -al /lib64/libcrypto*.so*
lrwxrwxrwx. 1 root root      19 Sep 26 06:25 /lib64/libcrypto.so -> libcrypto.so.1.0.2j
lrwxrwxrwx. 1 root root      19 Sep 26 06:25 /lib64/libcrypto.so.10 -> libcrypto.so.1.0.2j
-rwxr-xr-x. 1 root root 2485176 Sep 26 06:25 /lib64/libcrypto.so.1.0.2j
```

Inside the `lib` I created symlinks, with this result:

``` shell
~/Downloads/fritzing-0.9.3b.linux.AMD64/ cd lib

~/Downloads/fritzing-0.9.3b.linux.AMD64/lib/ sudo ln -s   libssl.so.1.0.0   

~/Downloads/fritzing-0.9.3b.linux.AMD64/lib/ ln -s /lib64/libssl.so.1.0.2j libssl.so.1.0.0

~/Downloads/fritzing-0.9.3b.linux.AMD64/lib/ ../Fritzing
/home/bob/Downloads/fritzing-0.9.3b.linux.AMD64/lib/Fritzing: error while loading shared libraries: libcrypto.so.1.0.0: cannot open shared object file: No such file or directory

~/Downloads/fritzing-0.9.3b.linux.AMD64/lib/ ln -s /lib64/libcrypto.so.1.0.2j libcrypto.so.1.0.0

~/Downloads/fritzing-0.9.3b.linux.AMD64/lib/ ../Fritzing                                        
/home/bob/Downloads/fritzing-0.9.3b.linux.AMD64/lib/Fritzing: /home/bob/Downloads/fritzing-0.9.3b.linux.AMD64/lib/libssl.so.1.0.0: version `OPENSSL_1.0.0' not found (required by /home/bob/Downloads/fritzing-0.9.3b.linux.AMD64/lib/libgit2.so.23)
/home/bob/Downloads/fritzing-0.9.3b.linux.AMD64/lib/Fritzing: /home/bob/Downloads/fritzing-0.9.3b.linux.AMD64/lib/libcrypto.so.1.0.0: version `OPENSSL_1.0.0' not found (required by /home/bob/Downloads/fritzing-0.9.3b.linux.AMD64/lib/libgit2.so.23)
```

I moved the symlinks from `lib` to `/lib64` but running `Fritzing` gives a similar message:

``` shell
~/Downloads/fritzing-0.9.3b.linux.AMD64/lib/ ../Fritzing 
/home/bob/Downloads/fritzing-0.9.3b.linux.AMD64/lib/Fritzing: /lib64/libssl.so.1.0.0: version `OPENSSL_1.0.0' not found (required by /home/bob/Downloads/fritzing-0.9.3b.linux.AMD64/lib/libgit2.so.23)
/home/bob/Downloads/fritzing-0.9.3b.linux.AMD64/lib/Fritzing: /lib64/libcrypto.so.1.0.0: version `OPENSSL_1.0.0' not found (required by /home/bob/Downloads/fritzing-0.9.3b.linux.AMD64/lib/libgit2.so.23)
```

Following the direction from @maweki on #3245, I removed `libgit2.so` with the same result. I removed `libgit2.so.23` and got this:

``` shell
~/Downloads/fritzing-0.9.3b.linux.AMD64/lib/ ../Fritzing     
/home/bob/Downloads/fritzing-0.9.3b.linux.AMD64/lib/Fritzing: error while loading shared libraries: libgit2.so.23: cannot open shared object file: No such file or directory
```

I created a symlink for the installed git library:

``` shell
ln -s /lib64/libgit2.so.0.24.3 libgit2.so.23
```

Which seems a little hinky, but it works. _**I'd like your opinion on that!**_ When I run `Fritzing` from the command line now, it opens the application, but with the following error:

``` shell
~/Downloads/fritzing-0.9.3b.linux.AMD64/lib/ ../Fritzing                                 
libpng warning: iCCP: known incorrect sRGB profile
```

and the window shows up expanded beyond the borders of my display (1920 X 1080) and looking "zoomed in." This feels like it's getting into a different issue, though. Any thoughts you have would be most welcome. Thanks!

P.S. Here's a screenie of what the window looks like when it opens. All the menus seem to work and I can zoom in and out on the workspace, but I have to drag the window around to be able to get at its different parts.

[image]

maweki, 2016-11-25:
I wouldn't think that the linked libraries have anything to do with the display bugs you're experiencing. Maybe some HiDpi issues or the window manager?

But yeah, Fritzing's dynamic linking policies are kinda f-ed and nobody seems to care.

mostlybob, 2016-11-26:
I don't know about the HiDPI thing - this is a pretty standard 1920 x 1080 display (Thinkpad T450s - Intel hd 5500). I'll keep at it.

mcarden, 2016-11-27:
I'm hitting the same error on Korora 24 and have now symlinked so many things to so many other things with no change in behavior that i wonder whether it's worth bothering. Something seems to be fundamentally broken.

AlanMcKay, 2017-01-16:
Trying the above on CentOS Linux release 7.2.1511, though of course the native library is older than the one for Fritzing.  It loads initially but then crashes after a few seconds.

[amckay@centos-gig lib]$ ls -al /lib64/libgit2.so*
lrwxrwxrwx. 1 root root     13 Jan 15 21:58 /lib64/libgit2.so -> libgit2.so.21
-rwxr-xr-x. 1 root root 834256 Mar 14  2015 /lib64/libgit2.so.0.21.4
lrwxrwxrwx. 1 root root     17 Jan 15 21:58 /lib64/libgit2.so.21 -> libgit2.so.0.21.4
[amckay@centos-gig lib]$ ln -s /lib64/libgit2.so.0.21.4 libgit2.so.23
[amckay@centos-gig lib]$ cd ..
[amckay@centos-gig fritzing-0.9.3b.linux.AMD64]$ ./Fritzing 
libpng warning: iCCP: known incorrect sRGB profile
/home/amckay/fritzing/fritzing-0.9.3b.linux.AMD64/lib/Fritzing: symbol lookup error: /home/amckay/fritzing/fritzing-0.9.3b.linux.AMD64/lib/Fritzing: undefined symbol: git_libgit2_init

KjellMorgenstern, 2019-07-12:
You can try the latests development build from Github Releases. Let me know if similar loading issues still exist.

closed 2019-07-12 by KjellMorgenstern