fritzing/fritzing-app #3293: Unable to update on mac OS Sierra 10.12.2 just after installing

dymbox, 2016-12-30:
After installing on Mac OS X Sierra 10.12.2 & checking for updates the following happens:

[image]

Then it downloads something and shows the following dialog

[image]

Then it closes itself, and on reopen says this

[image]

Then I'm trying to update it again thru Help -> Check for updates, and it shows me this

[image]

Finally, if I follow its advice and gonna try to rebuild the parts database, it says this

[image]

No manual update attempts performed from my side, just dragged it from .dmg to /Applications and started as brand new app.

mw9, 2016-12-31:
I have the same problem with OSX El Capitan 10.11.6.

Got this message box when initially updating:

[image]

Thereafter I experience much the same as the OP.

VeitSanner, 2017-01-13:
Same for me. Let's **up-vote** this issue.

hunter-nl, 2017-02-13:
I have also exactly the same problem on MacSierra 10.12.3. When can we expect a fix for this issue?

wez, 2017-02-19:
I'm seeing the same thing.  https://github.com/fritzing/fritzing-app/issues/3308 is a duplicate of this issue and some folks have reported that chmod'ing the $TMPDIR fixes it; it does not have any effect for me (the permissions on TMPDIR are correct, and honestly, suggesting making it 777 is very dangerous advice).

I've tried running Fritzing under dtruss but it doesn't show any failed syscalls relating to opening files or making directories, so I'm stumped as to what exactly it is failing on. :-(  Best thing to do as a workaround is just not update the parts library AFAICT.

gt500sc, 2017-02-19:
This worked for me:
http://superuser.com/questions/1133065/unable-to-write-to-tmp-osx-unable-to-correct-permissions-operation-not-permitt

VeitSanner, 2017-03-07:
This issue has been tagged with 'priority-critical' but work does not seem to have been started. The same applies to the duplicate #3308.

As @wez mentions modifying the permissions for $TMPDIR is not the best idea. Especially Apple could just reset permissions to default permissions with the next macOS update.

When can we expect a fix for the issue?

CC: @aknoerig

aknoerig, 2017-12-18:
Please try the fix that was just merged (see #3308). Hope this works for all of you!

closed 2017-12-18 by aknoerig