fritzing/fritzing-app #3302: Provide Fritzing AppImage for Linux

probonopd, 2017-01-14:
Already today, many Linux users download the Fritzing app directly from the Fritzing website rather use what comes with the Linux distribution, since the upstream-provided version is up-to-date while the version in the distributions (especially in stable/LTS versions) is outdated more often than not.

Providing an [AppImage](http://appimage.org/) would have, among others, these advantages:
- One app = one file = super simple for users: just download one AppImage file, [make it executable](http://discourse.appimage.org/t/how-to-make-an-appimage-executable/80), and run
- No unpacking or installation necessary
- No root needed
- No system libraries changed
- Just one format for all major distributions
- Works out of the box, no installation of runtimes needed
- Optional(!) desktop integration with `appimaged`
- Binary delta updates, e.g., for continuous builds (only download the binary diff) using AppImageUpdate
- Can GPG2-sign your AppImages (inside the file)

[Here is an overview](https://github.com/probonopd/AppImageKit/wiki/AppImages) of projects that are already distributing upstream-provided, official AppImages.

Do you think this would be useful?
I'd be happy to help producing an official Fritzing AppImage.

TheAssassin, 2019-01-18:
The following script builds a basic AppImage. Unfortunately, Fritzing hardcodes some paths to the build location, which is a problem, as AppImages must be relocatable. Are there any flags that can be set?

In any case, you should build your applications to load stuff relative to the main binary. That'll work reliably when an app is installed, but also in portable bundles like AppImages. Qt makes this quite easy even.

```sh
#! /bin/bash

set -x
set -e

# building in temporary directory to keep system clean
# use RAM disk if possible (as in: not building on CI system like Travis, and RAM disk is available)
if [ "$CI" == "" ] && [ -d /dev/shm ]; then
    TEMP_BASE=/dev/shm
else
    TEMP_BASE=/tmp
fi

BUILD_DIR=$(mktemp -d -p "$TEMP_BASE" fritzing-build-XXXXXX)

# make sure to clean up build dir, even if errors occur
cleanup () {
    if [ -d "$BUILD_DIR" ]; then
        rm -rf "$BUILD_DIR"
    fi
}
trap cleanup EXIT

# store repo root as variable
REPO_ROOT=$(readlink -f $(dirname "$0"))
OLD_CWD=$(readlink -f .)

# switch to build dir
pushd "$BUILD_DIR"

# configure build files with qmake
# we need to explicitly set the install prefix, as CMake's default is /usr/local for some reason...
qmake "$REPO_ROOT"

# build project and install files into AppDir
make -j$(nproc)
make install INSTALL_ROOT=AppDir

# now, build AppImage using linuxdeploy and linuxdeploy-plugin-qt
# download linuxdeploy and its Qt plugin
wget https://github.com/linuxdeploy/linuxdeploy/releases/download/continuous/linuxdeploy-x86_64.AppImage
wget https://github.com/linuxdeploy/linuxdeploy-plugin-qt/releases/download/continuous/linuxdeploy-plugin-qt-x86_64.AppImage

# make them executable
chmod +x linuxdeploy*.AppImage

# initialize AppDir, bundle shared libraries, add desktop file and icon, use Qt plugin to bundle additional resources, and build AppImage, all in one command
./linuxdeploy-x86_64.AppImage --appdir AppDir --plugin qt --output appimage

# move built AppImage back into original CWD
mv Fritzing*.AppImage "$OLD_CWD"
```

probonopd, 2019-01-18:
I had done a test AppImage in the past. It is still available:
https://bintray.com/probono/AppImages/Fritzing#files

probonopd, 2019-01-18:
And looking at http://fritzing.org/download/?donation=0 there is still no newer version than 0.9.3b, which was released on June 2, 2016. I wonder what happened?!

Morganlej, 2021-08-23:
Wow, i see there is now an AppImage of 0.9.8, great :)

Feedback running on Mageia 8, 64 bit, Plasma

Good: when launched it looks full OK.

But unusable: Upon start there is always an error message:
```
Regenerating parts database
Unable to find parts git repository
```
And then, probably caused by that there was no database generated:
```
Cannot read file screw_terminal_2_3.5mm.fzp:
No such file or directory.
```
And then
`Unable to find the following 159 part(s):`
-and all parts listed.

Then program launches, but only 6 parts are listed in core parts cathegory, and trying any other cathegory generate an error message similar to last one above.

I have not tried further.

KjellMorgenstern, 2021-08-25:
We have installed a fresh Mageia 8, 64 bit, Plasma and tested the AppImage.
It runs flawless.
[image]

TheAssassin, 2021-08-25:
Where's the build infrastructure located?

Morganlej, 2021-08-25:
Thank you Kjell for that test.
I also see you have updated the Mageia installation, good.

We have to track down why does it not work when i test on my production system...
Do we use same version?

[image]

KjellMorgenstern, 2021-08-25:
sha256 and filename:

`478f3ae8122aec5fbc077735b0e376b2cb621c9452b6d02104d1458765ab265f  fritzing-0.9.8-l92-b3d4fefa.AppImage`

Morganlej, 2021-08-25:
Match.

The error "Unable to find parts git repository" is it from trying internet or filesystem?

benreu, 2021-09-25:
I'm having the same problem with Fritzing 0.9.8 on Linux Mint 19. I can help with testing...

KjellMorgenstern, 2021-09-25:
We are providing an AppImage now, so I will close this issue.
If there are any specific issues with our AppImage, please open a distinct issue.

In the specific case above, i'd try to check if there are side effects ftom a previous version of Fritzing (there shouldn't, but its much harder to test)

closed 2021-09-25 by KjellMorgenstern

probonopd, 2021-09-25:
Hello, I'd like to run the AppImage through our automated test and add it to https://appimage.github.io/.
Is there a direct download URL to the AppImage?