fritzing/fritzing-app #3304: Freezes under Windows 10 Pro

CradyCopter, 2017-01-17:
I give up.  Even tried compatibility mode.

Installed in Windows 10 Pro and as soon as the UI loads it freezes solid.

dcconn, 2017-01-25:
This happens to me also.

dcconn, 2017-01-25:
I think I know what the problem was - I found a modal window hidden behind the main window asking about removing files prior to update. This windows required an answer. I was using task view and saw it there I've included a screenshot of the modal window after I clicked OK.
[image]

Once I allowed update to finish, I restarted fritzing and it seems ok :).

Regards,
David.

nopdotcom, 2017-07-22:
Windows 10 is surprisingly full of “modal dialog behind window” problems.

KjellMorgenstern, 2021-07-13:
No "pop unders" were reported since a long time. Please open a new ticket with exact Operating System and Fritzing version information if you encounter this.

closed 2021-07-13 by KjellMorgenstern