fritzing/fritzing-app #3308: Parts Update Results In "Unable to open temporary file"

ghoss, 2017-01-31:
Version **0.9.3**, fresh install on **Mac OSX 10.9.5**

When I select _Help > Check for updates_, I get informed that a new parts library update is available. Fritzing then displays a progress window "Downloading new parts", and once finished, I get this error message:

    Sorry, unable to install new parts: Unable to open temporary file
    Fritzing must nevertheless be restarted, so the "Close" button will close Fritzing.

Upon restarting Fritzing, I get this error:

    Unable to find the following 13 parts:
    '4001_4_x_2_input_NOR_gate_single' at ''
    'JST-B3P-VH' at 'JST-B3P-VH.fzp'
    '4070_4_x_2_input_XOR_gate_multipart' at ''
    … (and so on)

The app will then continue to start up normally. If I select _Help > Check for updates_ again, this time I get the error:

    The parts folder '/Applications/Fritzing.app/Contents/MacOS/fritzing-parts' has been changed (4)
    Fritzing cannot update the parts folder automatically.
    If you used to git to make changes, use git to update and then regenerate the database by
    selecting Part > Regenerate parts database…
    If you did not use git to make changes, then you may need to reinstall Fritzing.

If I select _Part > Regenerate parts database_, I get the error

    Unable to open temporary file.

Any ideas?

tempidoro, 2017-02-18:
This is exactly also what I experienced. I need help

tempidoro, 2017-02-18:
I applied the suggestion given at
http://stackoverflow.com/questions/22661459/osx-with-xampp-imagegd2-unable-to-open-temporary-file
within another problem discussion.
it worked for me also in our case

ghoss, 2017-02-18:
@tempidoro Thanks for your help, and I can confirm that the [workaround at stackoverflow.com](http://stackoverflow.com/questions/22661459/osx-with-xampp-imagegd2-unable-to-open-temporary-file) indeed solves the problem for me too.

In short, open a shell in Terminal and type the command

`chmod 777 ${TMPDIR}`

Teddy68, 2017-03-28:
Hi all,

I had the same problem in Mountain Lion.
I first installed the Fritzing application in another folder than "Applications", because already over-crowded. At this point, I found no way to fix the problem.

After reading this : 
> The parts folder '/Applications/Fritzing.app/Contents/MacOS/fritzing-parts' has been changed (4)

which is definitely not the place of my installation, I imagine Fritzing to be hard-coded to this location and not able to work elsewhere.
Then dropped Fritzing in the regular "Applications" folder, and there works like a charm.

Hope this helps

joshbg2k, 2017-04-19:
I currently have this issue and neither of the above fixes has worked.

joeljacobs, 2017-05-09:
I have tried both fixes as well and have the same problem on MacOS Sierra.

joeljacobs, 2017-05-09:
I found a solution. It looks like the QT temporary file call is misbehaving on a Mac. Instead of using the correct temporary folder, it uses / when launched from finder. Seems to be a bug in QT:

https://bugreports.qt.io/browse/QTBUG-56807

I was able to get mine to update by opening Fritzing from the command-line: 
/Applications/Fritzing.app/Contents/MacOS/Fritzing

It then successfully updated.

nopdotcom, 2017-07-20:
Say, what is a new binary release of fritzing blocking on?

esaheiskanen, 2017-08-30:
@joeljacobs solution did not work for me, mut when start fritzing from command-line with sudo it works. Not pretty, but it works.
`sudo /Applications/Fritzing.app/Contents/MacOS/Fritzing`

HartmannNico, 2017-09-25:
I did this: 
> Copy Fritzing from .dmg to desktop folder (uaaah!).
> Run the tool from desktop
> Update works fine. Restart works fine.
> Move everything to /Applications
Awful but effective.
However: 'Regenerate parts database...' will still bring up 'Unable to open temporary file'.

i0ntempest, 2017-11-23:
+1 on this issue. Running the app using Terminal solved the problem

securelpb, 2017-11-24:
I can confirm that running from command line per joeljacobs's suggestion works. No sudo required.

dlemper, 2017-12-14:
Seems like Fritzing is using current working directory as a temp dir on macOS.
When you open the application via Finder it gets '/' as the current working directory and there Fritzing doesn't have enough rights to create the temporary database for the parts.

For me it just worked like @securelpb wrote: run Fritzing from Terminal without sudo like this
`$ /Applications/Fritzing.app/Contents/MacOS/Fritzing`

dlemper, 2017-12-14:
OK, I had a look at the source code and the bug seems to be the following:
In this line https://github.com/fritzing/fritzing-app/blob/819abae0d75e85075275645a1ba5424fd2aea634/src/fapplication.cpp#L314
the temp file gets created under the current working directory as the docs of Qt tell you here
http://doc.qt.io/qt-5/qtemporaryfile.html#QTemporaryFile-1
This behaviour could be fixed with one of the following solutions:
1. create a QTemporaryFile without any parameter (`QTemporaryFile file();`) as it will create a temp file in the temp dir of the operating system using random characters, but then it won't have the '.db' file extension (which isn't necessary in my opinion)
2. add `QDir::tempPath()` to the filename (`QTemporaryFile file(QString("%1/XXXXXX.db").arg(QDir::tempPath()));`).

aknoerig, 2017-12-18:
Nice catch, thanks! It's now merged, hope this solution can be confirmed by everybody.

closed 2017-12-18 by aknoerig

tempidoro, 2017-12-27:
Hi,
I suppose that "It's now merged" from aknoering means that downloading Fritzing and updating Fritzing parts should work now. Sorry for this. I'm not expert about this stuff.
At the moment I copied Fritzing parts from the fritzing-parts-master and I substitute this parts to the one already in App folder.
Fritzing now works with the parts I tried to update. 
However, after the first working session, it do not work anymore when I reopen it.
If I delete the file parts.db before the new fritzing opening, it works again.
At moment I'm using Fritzing in this unconfortable way.

Intrinsically-Sublime, 2017-12-27:
@tempidoro that fix will not be available to you until the next release of Fritzing unless you compile it from source.

echristopherson, 2018-03-16:
I get the same thing on Windows 10 x64 (and yes, I had it on OS X before that too). I tried both
- giving myself full access recursively to the fritzing-parts folder; and
- changing the working directory of the Start Menu shortcut for Fritzing to my user temp directory.
Neither one made any difference.

As an aside, the Help > About menu item says my version is "Version 0.9.3 (b5c895d327c44a3114e5fcc9d8260daf0cbb52806 2016-04-19) 64 [Qt 5.6.1]", but the installer was for 0.9.3b and I can't find a commit b5c895d327c44a3114e5fcc9d8260daf0cbb52806 anywhere in the repository.

wcoburn, 2018-03-23:
@Intrinsically-Sublime Do you know if there's a plan for another official release, and if so when it might happen? The version on the fritizing.org website dates back to June 2016.

Intrinsically-Sublime, 2018-03-23:
I am only a user but I do spend a lot of time on the Fritzing forum and from what I can gather the main developers are working on porting over the code base from QT to web technologies resulting in little work being done to the main application. All of this is only speculation and should not be taken as fact. The only one that could tell us exactly when and how the next version will be released is @aknoerig.

I am running the testing branch of my fork on a daily basis and I can say it does fix a few things while introducing new bugs. Things like popup dialogs are no longer being brought to the front so you don't even know they are there unless you minimize the application.

artob, 2018-11-17:
Just ran into this as well with a new install of 0.9.3 on macOS 10.14.1. Cutting a new release would be very welcome indeed.

rcarmo, 2019-01-27:
Hi guys, it's 2019 and this is still an issue with the official 0.9.3b download, which is the only file linked from the downloads page. Is there _any_ hope of this getting fixed with an official build?